vsch/CLionArduinoPlugin #3: esp8266

shelkovkin, 2018-12-05:
I'm not sure if this should work. But I tried to configure to work with esp8266. By changing the board.txt to the one that comes with "https://github.com/esp8266/Arduino". As a result, I was able to create a new project for the desired board (by selecting from the list when creating it). But in the build configuration window, the list of boards remained the same, and the project is loaded with an error. It would be great if it would be possible to work with esp8266.
[image]
[image]
[image]
[image]
[image]

vsch, 2018-12-05:
@shelkovkin, I have not tried it so it probably does not work. The original intention for the customized boards.txt path to point it to a newer Arduino one. In this case it is a third party boards.txt file.

I will have to play with it and looks like the primitive parser in the plugin is not parsing the third party boards.txt correctly. I will try it out. There has to be a way to make it work with third party boards.

vsch, 2018-12-11:
@shelkovkin, I will have to add third party specific boards project handling. As I got myself a few esp8266 arduinos I have ability to test the full cycle. The third party boards.txt has different settings options and these need to be handled with that in mind.

Janrupf, 2019-01-11:
I also would like to have the ability to use esp8266 so I can use CLion for my projects.

PNixx, 2019-03-13:
And me to

SingingBush, 2019-04-14:
Any update on this? I'd like to deploy to a WeMos D1 Mini from CLion.

Janrupf, 2019-04-14:
Well, for now I would say use platform io, works pretty well and doesn't require any plugins

dwalker-uk, 2019-11-02:
Did anyone manage to get this working, i.e. ESP8266-based Arduino code in CLion?  I have the same issue as the originator of this thread, i.e. I can get the right boards.txt to appear when I create the project, but doesn't seem to be anywhere to update the list of boards that is used in the Build Settings.

Janrupf, 2019-11-02:
Yeah, you can easily use platform io as I mentioned (it effectively is Arduino)