vsch/CLionArduinoPlugin #8: CMake Syntax Error on New Project

onlineth, 2019-03-01:
When creating a new project for the Arduino Uno, I am met with these errors in the cmake debugger,

```
CMake Error at CMakeLists.txt:5 (set):
  Syntax error in cmake code at

    C:/.../CMakeLists.txt:5

  when parsing string

    ${<@PROJECT_NAME@>}_BOARD

  syntax error, unexpected cal_SYMBOL, expecting } (4)

-- Configuring incomplete, errors occurred!

[Finished]

gitcoinbot, 2019-03-03:
Issue Status: 1. **Open** 2. Started 3. Submitted 4. Done 

<hr>

__This issue now has a funding of 0.038 ETH (5.04 USD @ $132.72/ETH)  attached to it.__

 * If you would like to work on this issue you can 'start work' [on the Gitcoin Issue Details page](https://gitcoin.co/issue/vsch/CLionArduinoPlugin/8/2519).
* Want to chip in? Add your own contribution [here](https://gitcoin.co/issue/vsch/CLionArduinoPlugin/8/2519).
* Questions? Checkout <a href='https://gitcoin.co/help'>Gitcoin Help</a> or the <a href='https://gitcoin.co/slack'>Gitcoin Slack</a>
* $41,941.73 more funded OSS Work available on the [Gitcoin Issue Explorer](https://gitcoin.co/explorer)

gitcoinbot, 2019-03-03:
Issue Status: 1. **Open** 2. Started 3. Submitted 4. Done 

<hr>

__Workers have applied to start work__.

These users each claimed they can complete the work by 6 days, 22 hours from now.
Please review their action plans below:

**1) [jacobcreech](https://gitcoin.co/profile/jacobcreech) has applied to start work _(Funders only: [approve worker](https://gitcoin.co/issue/vsch/CLionArduinoPlugin/8/2519?mutate_worker_action=approve&worker=jacobcreech) | [reject worker](https://gitcoin.co/issue/vsch/CLionArduinoPlugin/8/2519?mutate_worker_action=reject&worker=jacobcreech))_.**

I have a few debugging questions. When compiling a new project, are you setting the project_name in the cmakeLists?

https://github.com/vsch/CLionArduinoPlugin/blob/master/resources/com/vladsch/clionarduinoplugin/templates/project/CMakeLists.txt#L6

You should be setting the project name here like in the test cases
https://github.com/vsch/CLionArduinoPlugin/blob/master/test-resources/created/sketch_1/CMakeLists.txt#L6
Also found here
https://github.com/vsch/CLionArduinoPlugin/blob/master/assets/images/Screenshot_ChangeBuildDiff.png

So the correct change would be to change 
https://github.com/vsch/CLionArduinoPlugin/blob/master/resources/com/vladsch/clionarduinoplugin/templates/project/CMakeLists.txt#L6
to 
set(${CMAKE_PROJECT_NAME}_BOARD pro)

Learn more [on the Gitcoin Issue Details page](https://gitcoin.co/issue/vsch/CLionArduinoPlugin/8/2519).

vsch, 2019-03-03:
@onlineth, I'll take a look at this. I am pretty certain it is caused by the convoluted settings implementation where it shows the selection in the dropdown if no valid value is provided by settings but does not actually change the settings. 

When the CMakeLists.txt is generated project settings values are used not what is shown in the dropdown.

vsch, 2019-03-03:
@onlineth, I prepared a new release with accumulated fixes, which I thought were minor. I tried creating a new sketch and new library projects from a new CLion 2018.3 installation and CLion 2019.1 EAP and both create valid projects which compile.

Can you please try the [CLionArduinoPlugin.1.4.6.zip] and let me know if it works for you.

[CLionArduinoPlugin.1.4.6.zip]: https://github.com/vsch/CLionArduinoPlugin/blob/master/dist/CLionArduinoPlugin.1.4.6.zip

[CLionArduinoPlugin.1.4.6.zip]: https://github.com/vsch/CLionArduinoPlugin/blob/master/dist/CLionArduinoPlugin.1.4.6.zip

onlineth, 2019-03-03:
@vsch Thank you for your response abd help. 

Using that version, I am still running into this issue - is there something I should be changing manually?

[image]

vsch, 2019-03-03:
@onlineth, not really. It looks like a bug that could be an OS/Path related issue. Let me try it under windows and I will let you know.

vsch, 2019-03-04:
@onlineth, I spent about 3 hours wrestling with Windows, actually the first hour was waiting for it to update my installation under Parallels Desktop.

I could not get CLion 2018.3 installation fully functional. Too many installation options and my Windows experience is rusty and I am good keeping it that way.

I was not sure which toolset needed to be installed for CLion to work with Arduino SDK tools installed by Arduino IDE or how to get CLion to pass the environment variable or be sure whether the Arduino.cmake expects `/` or `\` for path separator. So needless to say I could not validate that the whole thing works.

However, the plugin did create the right CMakeLists.txt and I was not able to duplicate your error. So my suspicion that it was path related  (Windows vs Unix) did not pan out. 

[image]

Working in Windows for me is torture. The keys are all wrong. None of the usual programs are available. Standard Mac OS/Linux programs must be installed and configured. 

Someone with a Windows system setup for development will have to give this a go. I cannot spend any more time on it.

Sorry.

onlineth, 2019-03-04:
@vsch Thank you so much for trying, if you want, I can email you remote credentials to a Windows environment loaded with CLion ready to go. If not I understand, Unix and Windows are a bit of a difference I'm sure for this kind fo development, I just don't want to pay $45 for https://www.visualmicro.com/

gitcoinbot, 2019-03-08:
Issue Status: 1. Open 2. **Cancelled** 

<hr>

__The funding of 0.038 ETH  (5.22 USD @ $137.41/ETH) attached to this issue has been **cancelled** by the bounty submitter__

 
* Questions? Checkout <a href='https://gitcoin.co/help'>Gitcoin Help</a> or the <a href='https://gitcoin.co/slack'>Gitcoin Slack</a>
* $39,955.57 more funded OSS Work available on the [Gitcoin Issue Explorer](https://gitcoin.co/explorer)

vsch, 2019-03-08:
@onlineth, I found instructions for Windows Arduino CMake installation when looking into #9 , so will be able to make progress on this issue this weekend.

I'll take some Prozac to keep Tourette's under control and work on Windows.

barkerest, 2019-07-15:
This may help you in your search for the bug.  I extracted the templates from the jar into a directory and I set the configuration to use the extracted templates.  With this configuration I get the expected blank sketch file and a working CMaleLists.txt file.

vsch, 2019-07-16:
@barkerest, Thank you. It does help narrow it down. It sound like an EOL difference between `\n` and `\r\n`, since windows expects `\r\n` and the jar has `\n'.

barkerest, 2019-07-17:
My extracted templates still have the Unix style `\n` line endings.

vsch, 2019-07-19:
@barkerest, good to know. I suspect it is something related to EOL and how they are processed when the template is streamed from the resource. The only way to debug this is for me to get Windows development setup because I am unable to duplicate this under OS X.

At least now there is a workaround that solves the issue.

onlineth, 2019-07-19:
FYI If you do want a Windows development setup, [this is free from Microsoft](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines)

vsch, 2019-07-19:
@onlineth, thanks. I will try it if I cannot get my Windows 10 under Parallels working trouble free.

RedHoodedWraith, 2019-07-20:
I am a Windows user and I would like to fix this issue but I don't know where to start nor what I should be doing. Where should I start in trying to solve the parsing issue or whatever is causing this issue? What am I looking for?

bowbahdoe, 2019-07-21:
https://github.com/vsch/CLionArduinoPlugin/blob/master/src/com/vladsch/clionarduinoplugin/generators/cmake/CMakeListsTxtBuilder.kt#L883
https://github.com/vsch/CLionArduinoPlugin/blob/master/src/com/vladsch/clionarduinoplugin/generators/cmake/CMakeListsTxtBuilder.kt#L970

These two lines seem to responsible for the `<@` and `@>` tags, but i its unclear what changes behaviour on the different OS's

RedHoodedWraith, 2019-07-21:
After some help from @bowbahdoe and a bit of trial and error, it seems that something in this file creation causes both the extra `<@``@>` tags comment `#`s on incorrect lines. After removing them and commenting the correct lines, does the sketch build. 

_Left_ is the version from Windows after fixing it and _Right_ is an example created on Windows:
[image]

somefunAgba, 2020-01-18:
This is the solution. The autogeneration of the CMakeLists.txt has to be fixed to prevent @<PROJECT_NAME>@ and generate PROJECT_NAME instead.

JedrzejczykRobert, 2020-01-29:
Problem is in TemplateResolver.kt (line 107):

name.startsWith(prefix) always return false, because prefix have '\' bur name have '/'

Logs form Windows 10:

Bundled Templates jar:file:/C:/Users/robert/.CLion2019.3/config/plugins/CLionArduinoPlugin/lib/CLionArduinoPlugin.jar!/com/vladsch/clionarduinoplugin/templates/ 
resource.path: file:/C:/Users/robert/.CLion2019.3/config/plugins/CLionArduinoPlugin/lib/CLionArduinoPlugin.jar!/com/vladsch/clionarduinoplugin/templates/ 
jarPath:       /C:/Users/robert/.CLion2019.3/config/plugins/CLionArduinoPlugin/lib/CLionArduinoPlugin.jar 
baseDir:       \com\vladsch\clionarduinoplugin\templates 
templateType:  project/sketch 
templateDir:   \com\vladsch\clionarduinoplugin\templates\project\sketch 
prefix:        com\vladsch\clionarduinoplugin\templates\ 

jarEntry.name: com/vladsch/clionarduinoplugin/templates/@@.cpp 
jarEntry.name: com/vladsch/clionarduinoplugin/templates/@@.ino 
jarEntry.name: com/vladsch/clionarduinoplugin/templates/@library_name@.cpp 
jarEntry.name: com/vladsch/clionarduinoplugin/templates/@library_name@.h 
jarEntry.name: com/vladsch/clionarduinoplugin/templates/library/@library_name@.cpp 
jarEntry.name: com/vladsch/clionarduinoplugin/templates/library/@library_name@.h 
jarEntry.name: com/vladsch/clionarduinoplugin/templates/project/CMakeLists.txt

JedrzejczykRobert, 2020-01-30:
Working patch, lines 92-99 in TemplateResolver.kt:

    if (resource.protocol == "jar" && templatesDir == null) {
        val jarPath = resource.path.substring(5, resource.path.indexOf("!")).replace("/",  File.separator)
        val jar = JarFile(URLDecoder.decode(jarPath, "UTF-8"))
        baseDir = File(resource.path.substring(5 + 1 + jarPath.length).replace("/",  File.separator))
        templateDir = baseDir + templateType

        val entries = jar.entries() //gives ALL entries in jar
        val prefix = baseDir.path.substring(1).replace("/",  File.separator) + File.separator

vsch, 2020-01-30:
@JedrzejczykRobert, thank you for the fix. I will incorporate it and release a new version today.

JedrzejczykRobert, 2020-01-30:
Another fix for get ARDUINO_SDK_PATH from ENV on Windows:

ArduinoToolchainFiles.java, after line 52 insert code:

    if (System.getProperty("os.name").toLowerCase().contains("win"))
    {
        String arduinoSdkPathEnv = System.getenv("ARDUINO_SDK_PATH").replace("\\", "/");
        String windowsPathsString =
            "set(ARDUINO_SDK_PATH \"" + arduinoSdkPathEnv + "\")" + "\r\n" +
            "message(\"ARDUINO_SDK_PATH ${ARDUINO_SDK_PATH}\")" + "\r\n";

        VirtualFile windowsPaths = platformDirectory.createChildData(this, "WindowsPaths.cmake");
        OutputStream windowsPathsOutputStream = windowsPaths.getOutputStream(this);
        InputStream windowsPathsInputStream = IOUtils.toInputStream(windowsPathsString);

        try {
            IOUtils.copy(windowsPathsInputStream, windowsPathsOutputStream);
        } finally {
            closeStreams(windowsPathsOutputStream, windowsPathsInputStream);
        }
    }

vsch, 2020-01-30:
@JedrzejczykRobert, thank you. I have a new build. Do you want to wait in case you find more issues or do you want to try the new build to see if it works? 

I can add it to the `dist` directory of this repo so you can try it to make sure the new build does not introduce some IDE version incompatibilities.

JedrzejczykRobert, 2020-01-30:
I can test your bilds, I use CLion 2019.3 on Win 10.

vsch, 2020-01-30:
@JedrzejczykRobert, thanks. I also tested it on 2019.3 on OS X.

The zip is available https://github.com/vsch/CLionArduinoPlugin/blob/master/dist/CLionArduinoPlugin.1.5.0.zip

Can be installed using "Install Plugin from Disk" in plugins config settings menu.

JedrzejczykRobert, 2020-01-30:
I'm sorry, i forgot about: line 105 in TemplateResolver.kt:

                    val name = jarEntry.name.replace("/",  File.separator)

Fix for get ARDUINO_SDK_PATH from ENV working OK.

JedrzejczykRobert, 2020-01-30:
cmake function print_programmer_list() not work, fix:

cmake\Platform\Arduino.cmake, line 698, should be:

    load_arduino_style_settings(${PLATFORM}_PROGRAMMERS "${PLATFORM_PATH}/programmers.txt")

I added at end of CMakeLists.txt file:

print_board_list()
print_programmer_list()

this show all boards and programmers supported by Arduino. 
Wihtout this fix programmers list is empty.

vsch, 2020-01-30:
@JedrzejczykRobert, I added the mods you mentioned and forked arduino cmake to make changes to Arduino.cmake.

New build in dist https://github.com/vsch/CLionArduinoPlugin/blob/master/dist/CLionArduinoPlugin.1.5.1.zip

JedrzejczykRobert, 2020-01-30:
I checked, it works, it's great. Thanks.

[image]

vsch, 2020-01-31:
@JedrzejczykRobert, I released a new version with the fixes and it should be available for download and update in the IDE after it is verified.

Thank you for identifying and resolving this issue. It is much appreciated.

JedrzejczykRobert, 2020-02-06:
[image]

closed 2020-02-13 by onlineth