vsch/CLionArduinoPlugin #9: Could not find Arduino SDK

Mikolaj98p, 2019-03-06:
Hello I have error:
```
CMake Error at cmake/ArduinoToolchain.cmake:82 (message):
  Could not find Arduino SDK (set ARDUINO_SDK_PATH)!
Call Stack (most recent call first):
  D:/programs/JetBrains/Toolbox/apps/CLion/ch-1/182.5107.50/bin/cmake/win/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:7 (project)
```

vsch, 2019-03-06:
@Mikolaj98p, it is the error I could not figure out when running on Windows for solving #8 

I do not use Windows and have no development machine for Windows. Doing it under Parallels Desktop on the Mac is an exercise in frustration. You should probably follow the Arduino CMake instructions https://github.com/francoiscampbell/arduino-cmake/tree/dbd4fe372b9a0120325b429b3ed23c311ab3dc66#installing-clion--arduino-ide--mingw for setting up a CLion Arduino Dev environment on Windows.

vsch, 2019-03-06:
@Mikolaj98p, if the instructions solve the problem then please let me know and I will add the link to the FAQ.

Mikolaj98p, 2019-03-06:
Now I have another error:
```
CMake Error at cmake/Platform/Arduino.cmake:2130 (message):
  ALL_SRCS not set: must define SRCS or SKETCH for target untitled7
Call Stack (most recent call first):
  cmake/Platform/Arduino.cmake:498 (required_variables)
  CMakeLists.txt:18 (generate_arduino_firmware)
```

vsch, 2019-03-06:
@Mikolaj98p, can you post your CMakeLists.txt file here, I think the same issue as #8 is showing up on your system.

I see you are using CLion 2018.2 and it does not support the full API used by the plugin. I thought I handled this in the code but it is possible that on Windows there are other differences that combine to expose a bug. 

Can you upgrade to CLion 2018.3? or 2018.2 is the only version you can use?

Mikolaj98p, 2019-03-06:
[CMakeLists.txt](https://github.com/vsch/CLionArduinoPlugin/files/2938204/CMakeLists.txt)

vsch, 2019-03-06:
The plugin is setting the SKETCH variable but for some reason leaving the line commented out.

If you uncomment line `set(${PROJECT_NAME}_SKETCH untitled7.ino)` as in the file below, it should work:

```text
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 98)
set(PROJECT_NAME untitled7)
set(${PROJECT_NAME}_BOARD nano)
set(ARDUINO_CPU atmega328old)
project(${PROJECT_NAME})
# set(${PROJECT_NAME}_SRCS)
# set(${PROJECT_NAME}_HDRS)
# set(${PROJECT_NAME}_LIBS)
# link_directories(${CMAKE_CURRENT_SOURCE_DIR}/)
# add_subdirectory()
set(${PROJECT_NAME}_SKETCH untitled7.ino)
# set(${PROJECT_NAME}_PROGRAMMER avrispmkii)
set(${PROJECT_NAME}_PORT COM4)
# set(${PROJECT_NAME}_AFLAGS -v)
# set(nano.upload.speed 9600)
generate_arduino_firmware(${PROJECT_NAME})
```

I will take a look at the code and see what is causing this.

I can also use the instructions for Windows setup to see if I can get my Windows virtual machine working with CLion.

dancingfrog, 2019-03-22:
Had the same problem in CLion 2018.3.4 on Windows 10. Uncommenting the set...sketch call fixed it!

AridTag, 2019-03-28:
Just to add more information to this issue and #8 for you @vsch 

For me using 2019.1 CMakeLists.txt generated like this
```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 17)
set(<@PROJECT_NAME@> untitled)
set(${<@PROJECT_NAME@>}_BOARD nano)
set(ARDUINO_CPU atmega328)
project(${<@PROJECT_NAME@>})
# set(${<@PROJECT_NAME@>}_SRCS)
# set(${<@PROJECT_NAME@>}_HDRS)
# set(${<@PROJECT_NAME@>}_LIBS)
# link_directories(${CMAKE_CURRENT_SOURCE_DIR}/)
# add_subdirectory()
# set(${<@PROJECT_NAME@>}_SKETCH untitled.ino)
set(${<@PROJECT_NAME@>}_PROGRAMMER avrispmkii)
# set(${<@PROJECT_NAME@>}_PORT /dev/cu.usbserial-00000000)
set(${<@PROJECT_NAME@>}_AFLAGS -v)
# set(nano.upload.speed 9600)
generate_arduino_firmware(${PROJECT_NAME})
```

However I had to modify it to this to get it to work
```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 17)
set(PROJECT_NAME untitled)
set(${PROJECT_NAME}_BOARD nano)
set(ARDUINO_CPU atmega328)
project(${PROJECT_NAME})
# set(${<@PROJECT_NAME@>}_SRCS)
# set(${<@PROJECT_NAME@>}_HDRS)
# set(${<@PROJECT_NAME@>}_LIBS)
# link_directories(${CMAKE_CURRENT_SOURCE_DIR}/)
# add_subdirectory()
set(${PROJECT_NAME}_SKETCH untitled.ino)
set(${PROJECT_NAME}_PROGRAMMER avrispmkii)
# set(${<@PROJECT_NAME@>}_PORT /dev/cu.usbserial-00000000)
set(${PROJECT_NAME}_AFLAGS -v)
# set(nano.upload.speed 9600)
generate_arduino_firmware(${PROJECT_NAME})
```

OscarVanL, 2019-04-02:
I'm having the same issue.

`"C:\Program Files\JetBrains\CLion 2019.1\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - MinGW Makefiles" C:\Users\Oscar\CLionProjects\Tesla2
CMake Error at cmake/ArduinoToolchain.cmake:82 (message):
  Could not find Arduino SDK (set ARDUINO_SDK_PATH)!
Call Stack (most recent call first):
  C:/Program Files/JetBrains/CLion 2019.1/bin/cmake/win/share/cmake-3.13/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:4 (project)`

My CMakeLists.txt is simple, and the setting of SKETCH is uncommented.

`cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME Tesla2)
project(${PROJECT_NAME})

set(${CMAKE_PROJECT_NAME}_SKETCH Tesla2.ino)

#### Uncomment below additional settings as needed.
set(${CMAKE_PROJECT_NAME}_BOARD mega)
# set(${CMAKE_PROJECT_NAME}_PORT /dev/ttyACM0)
# set(mega.build.mcu atmega2560)
# set(mega.upload.protocol wiring)
# set(mega.upload.speed 115200)

generate_arduino_firmware(${CMAKE_PROJECT_NAME})
`

I'm running it on Windows.

OscarVanL, 2019-04-03:
Nevermind, the above issue went away by itself after reopening CLion and allowing CMake to re-run.

vsch, 2019-04-04:
@OscarVanL, the plugin does try to force CLion to reload CMakeLists.txt after creation. Something is causing failure to do so.

Do you have any exceptions in CLion?

I will try to get my Windows environment setup so I can debug on Windows. It seems to be an OS specific issue.

OscarVanL, 2019-04-05:
Hi vsch, I do get an error from CLion.

"Error running 'upload': Cannot run program "C:\Users\OscarVanL\*******************\EDU\cmake-build-debug\EDU.elf" (in directory "C:\Users\OscarVanL\*******************\EDU\cmake-build-debug"): CreateProcess error=193, %1 is not a valid Win32 application"

If I copy and paste my .ino source code and put it into the official Arduino application, it compiles and runs on my Arduino fine.

MarcoStalder, 2019-05-01:
> Just to add more information to this issue and #8 for you @vsch
>
> For me using 2019.1 CMakeLists.txt generated like this
>
> ```
> cmake_minimum_required(VERSION 2.8.4)
> set(CMAKE_TOOLCHAIN_FILE [...]

I had the same situation and changed the CMakeLists.txt the same way as above (getting rid of the <@ and @> and uncommenting set...SKETCH). Then I get this error:

CMake Error at cmake/Platform/Arduino.cmake:1829 (message):
  Sketch does not exist:
  C:/Users/[...]/test_11/test_11.ino

So I tried to create a new Arduino Sketch File from within CLion. Therefore, in the Project tab, I right-clicked on the folder test_11 -> New -> Arduino Sketch. Create New Sketck File dialog box opens, I enter test_11.ino (also tried just test_11), hit OK. Create New Sketck File dialog box closes. Nothing happens. Can't find any such file anywhere.

So I created a new Arduino Sketch File test_11.ino from the Arduino IDE and copied it to the test_11 folder.

Now reloading CMakeList.txt works without error. (except for "Some source files are located outside of CMakeLists.txt directory. You can change the project root or ignore this in future.")

But when I try to Run test_11 from CLion, I get:

[ 89%] Built target mega_CORE
Scanning dependencies of target test_11
[ 92%] Building CXX object CMakeFiles/test_11.dir/test_11_test_11.ino.cpp.obj
In file included from C:/Users/[...]/test_11/cmake-build-debug/test_11_test_11.ino.cpp:8:0:
C:/Program Files (x86)/Arduino/hardware/arduino/avr/cores/arduino/Arduino.h:153:16: error: conflicting declaration of 'void setup()' with 'C' linkage
 void setup(void);
                ^
C:/Users/Marco/Documents/ETH/Master/Masterarbeit/Software/Arduino/test_11/test_11.ino:1:6: note: previous declaration with 'C++' linkage
 void setup() {
      ^
mingw32-make.exe[3]: *** [CMakeFiles/test_11.dir/test_11_test_11.ino.cpp.obj] Error 1
CMakeFiles\test_11.dir\build.make:65: recipe for target 'CMakeFiles/test_11.dir/test_11_test_11.ino.cpp.obj' failed
mingw32-make.exe[2]: *** [CMakeFiles/test_11.dir/all] Error 2
CMakeFiles\Makefile2:140: recipe for target 'CMakeFiles/test_11.dir/all' failed
CMakeFiles\Makefile2:152: recipe for target 'CMakeFiles/test_11.dir/rule' failed
Makefile:143: recipe for target 'test_11' failed
mingw32-make.exe[1]: *** [CMakeFiles/test_11.dir/rule] Error 2
mingw32-make.exe: *** [test_11] Error 2

(the test_11.ino file is just the standard one that gets created by the Arduino IDE:
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
)

Any idea on what else I can try?

MarcoStalder, 2019-05-01:
In the meantime I could resolve the above issue by adding #include "Arduino.h" to the test_11.ino file. (Sorry for not trying this before.)
But now I just get the same error as stated here:

> Hi vsch, I do get an error from CLion.
>
> "Error running 'upload': Cannot run program "C:\Users\OscarVanL*******************\EDU\cmake-build-debug\EDU.elf" (in directory [...]

@OscarVanL have you solved this somehow?

vsch, 2019-05-01:
@MarcoStalder, the project does not create Windows executables. You need to select a run configuration with name ending in "updload" and build it. This will result in the project being built and uploaded using the defined programmer.

MarcoStalder, 2019-05-01:
@vsch, thank you for the prompt reply! That solved it. I kind of tried to do this before, but back then, since it didn't work with the preset settings, I changed the Target to mega_CORE (I intuitively thought that would make it run on the Arduino Mega). Instead I should have left the Target and just selected test_11 as the Executable, which I did now.

MarcoStalder, 2019-05-01:
@vsch, I now created a list on what made it work for me (on Windows 10 with CLion 2019.1.2). This was originaly meant to be a crutch for my future self. But I figured, it might as well help others in my situation. Feel free to use and spread it in whatever form you like.

about Aduino Support Plugin for CLion:
	* maybe you need to install MinGW (see https://github.com/francoiscampbell/arduino-cmake/tree/dbd4fe372b9a0120325b429b3ed23c311ab3dc66#installing-clion--arduino-ide--mingw)
	* to create a new project:
		* New Project -> Arduino Sketch
			* the right Settings can be read from the settings of the Arduino IDE (if the Arduino IDE is already working with the Arduino at hand) => hit Create
		* in CMakeList.txt:
			* replace all <@PROJECT_NAME@> with PROJECT_NAME
			* uncomment set(${PROJECT_NAME}_SKETCH ...)
		* create a .ino-file with the same name as the project (or rather the name shown in set(${PROJECT_NAME}_SKETCH ...) in CMakeList.txt)
			* I could not do this from within CLion => I created this file from the Arduino IDE and then copied it to the right location (the project folder of this project)
				* (if you save it directly from the Arduino IDE, it will be within an additional folder => get it out of there and delete the remaining empty folder)
		* insert #include "Arduino.h" as the first line of the above mentioned .ino-file
		* Reload CMakeList.txt
			* e.g. through File -> Reload CMake Project
		* at the top right of the screen select one of the Run/Debug Configurations ending with upload => hit Run(-triangle) => set Executable to the one in the list with the same name as the project => hit Run
			* (((the error message about Win32-Anwendung/executable is not relevant (it is not meant to run on Windows, but only on the Arduino) (Error running 'bms_master_arduino-upload': Cannot run program "C:\Users\...\cmake-build-debug\bms_master_arduino.elf" (in directory "C:\Users\...\cmake-build-debug"): CreateProcess error=193, %1 ist keine zulässige Win32-Anwendung))))
	* Serial Monitor
		* e.g. to see if the upload really succeeded and the Sketch is runing on the Arduino
		* on the bottom left of the screen: hit Arduino Support -> hit Edit Settings (screw-wrench) -> set the Port to the same as when creating the project (can also be found in CMakeList.txt), select the right Baudrate (the one you use in the Sketch in Serial.begin(...)) -> hit OK
		* hit Connect COM... @ ... (e.g. COM13 @ 9600 (Port Number and Baudrate))
			* to disconnect hit the same button (now labeled Disconnect COM... @ ...)

AgentRev, 2019-07-12:
In relation to the initial issue, I repeatedly got the error, and the problem was that it wasn't detecting my Windows Store version of the Arduino IDE, so I uninstalled it and downloaded the native Windows installer version instead, now it seems to detect it. I also experienced #8 on CLion 2019.1.

dancingfrog, 2019-07-13:
Same here... I was using the Windows store version of Arduino IDE. Guess we
need to download/install direct from Arduino.

Peace & Love,
John