vsch/CLionArduinoPlugin #10: Impossible to add external library

ksamp666, 2019-03-10:
Uncommenting "link_directories(${CMAKE_CURRENT_SOURCE_DIR})" in cmake file does not add a new external library to project. Is it possible to add them at all?

vsch, 2019-03-10:
@ksamp666, depends on what you mean by external library. In the Arduino CMake, the external libraries are ones where the directory and the contained library files have the same name.

`link_directories` gives a list of parent directories containing arduino libraries.

For example and short explanation:

```
# For this example (libs will contain additional arduino libraries)
# An Arduino library my_lib will contain files in libs/my_lib/: my_lib.h, my_lib.cpp + any other cpp files
link_directories(${CMAKE_CURRENT_SOURCE_DIR}/lib ${CMAKE_CURRENT_SOURCE_DIR}/util)
```

So if you have libraries you want to include with your project add their containing directory absolute path to `link_directories()` and Arduino CMake will include them in the build

A bit weird of an implementation but I guess this is how it is implemented.

ksamp666, 2019-03-10:
@vsch Thank you very much for your explanation! It helped!

closed 2019-03-10 by ksamp666

vsch, 2019-03-10:
@ksamp666, you can add the additional directories to the project build settings   
[image]

Select the additional library directory and add your libraries, separated by `;`. Currently it expects all directories to be relative to project directory but in the code it would be an easy fix to make it leave absolute paths alone and only prefix current project directory to relative paths.

[ArduinoCMakeListsTxtBuilder.kt: Line 213](https://github.com/vsch/CLionArduinoPlugin/blob/master/src/com/vladsch/clionarduinoplugin/generators/cmake/ArduinoCMakeListsTxtBuilder.kt#L213)