vsch/CLionArduinoPlugin #11: Clion Code menu options are grayed out

fbrier, 2019-03-11:
I installed the CLionArduinoPlugin into CLion 2018.3.4. Created a new Arduino Sketch project. Added a C++ class under the project folder (which contains the CMakeLists.txt) file. Positioned the cursor at the end of the .cpp file. Selected the "Code" menu. The top 4 menu items are grayed, from "Override Functions" to "Generate...".

Creating a non-Arduino, C++ executable, project and performing the same operations, has the menu options active (not grayed), and selecting "Generate..." allows you to successfully create constructors and destructors. Is there a step I am missing? Thank you.

vsch, 2019-03-11:
[image]

I only saw the menu get disabled until the CMakeLists.txt was reloaded and the .ino file was showing as not being part of any dependency.

Can you try doing Reload on CMakeLists.txt to see if it clears this condition.

fbrier, 2019-03-12:
If I select the .ino file, the menu options, as you described, are not disabled. But if I select the generated .cpp or header file, they are disabled. There is a message at the top of the editor of the .cpp and header files - "This file does not belong to any project target, code insight features might not work properly." I did uncomment the CMakeLists.txt line and add the source file:

```set(${PROJECT_NAME}_SRCS TestingGenerate.cpp)```

and added the header file with:

```set(${PROJECT_NAME}_HDRS User_Setup.h TestingGenerate.h)```

However, this did not remove the message, nor make the menu options active. My understanding was the goal of your plugin was not to code using the .ino files, but leverage CLion's advanced features for C++. Isn't the .ino file supposed to be thin, with the real functionality contained in normal C++ files? Thank you for your help.

vsch, 2019-03-12:
@fbrier, the files do not become part of the project until CMakeLists.txt is reloaded (ie. run with CMake) to generate new make files with new dependencies. This is how CLion works.

You should either set CMakeLists.txt to auto-load or manually select reload from the context menu.

fbrier, 2019-03-13:
I changed the CMake to auto-load, but that did not fix the problem, although not setting that might have led to another issue. The fix appears to be right clicking the folder containing the sources (User_Setup.h and <projectname>.ino) in the Project panel, and selecting the "Mark Directory as" and then include the folder as containing project sources. Now the menu options are not grayed. Is this the answer? Or is there a better way? Thank you.

vsch, 2019-03-13:
@fbrier, this appears to be new behaviour for 2018.3. I tried a project which did not have any source directory roots set and it needed this setting. This was not the case before.

I will add this to the to do list but have to be up front that it will take a while for me to get to working on this plugin. I am overloaded at the moment and probably the condition will remain the same for the next month or two.

fbrier, 2019-03-13:
No problem. I totally understand. Thank you for doing what you have. The whole point of open source is to have your fellow developers chip in and push the ball further down the field than you could alone. So assuming I have time as well, perhaps I can help attack your wish list, and if I find something I would like, add it. Again, thank you.

vsch, 2019-03-13:
@fbrier, I could not have said it better myself or put it into more diplomatic terms.

I had about three weeks over end year holidays for Arduino projects and spent most of it on this plugin. I am glad I did because I love JetBrains IDEs and find the Arduino IDE a toy in comparison, but I did run out of time and now have my Arduino projects undone and waiting for the next break in my schedule.

I absolutely welcome everyone adding to the effort. Especially when they are working on Windows and the issues are OS specific. I only have Windows setup under Parallels Desktop for OS X and resort to it for compatibility testing in extreme circumstances.

I don't do Arduino, or CLion or CMake based development professionally or have much experience with any of them, so I don't have more answers than anyone else doing a search for them on the Web.

As an aside on Windows: 

I was an avid professional Windows developer from Windows 3.1 all the way to Vista. I got fed up of Microsoft making clueless changes to "Improve user friendliness" of the OS for retired grandmothers while ignoring what it entailed for existing power users.

I switched to Mac development 10+ years ago. At first it was hard to get over old habits but once I got used to Unix way I realized what I was missing in Windows. It is really a more consistent and orthogonal OS implementation and OS X adheres to consistency religiously.

Now that I have Windows out of my system, going back triggers severe bouts of Tourette's. 

I cannot stand the loss of two hours for windows updates. 

I find the changes to where settings are in the latest OS a PITA to follow. I know the setting I need to tweak, I just don't know where Microsoft has decided to stash it in the latest incarnation of the OS. 

Dev tools and utilities have better compatibility with OS X and Unix than Windows. Having to figure out what I need to install under Windows is a journey all in itself fraught with clickbait sites, download site wannabe's and generally a PITA experience.

Not a good set of motivational factors for something that is essentially a hobby effort for pleasure.