vsch/CLionArduinoPlugin #12: Invalid Arduino Board ID

l0rn, 2019-04-28:
Hello,

thanks for the great work integrating my favorite IDE with arduino! Unfortunately I'm having trouble setting it up correctly. I created a fresh project which immediately throws the error above. I'm using the board ID `nano` but it actually errors out on any board variant.

The `CMakeLists.txt` generated by this plugin:

```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 17)
set(PROJECT_NAME untitled1)

set(${PROJECT_NAME}_BOARD nano)
set(ARDUINO_CPU atmega328old)
project(${PROJECT_NAME})

# Define additional source and header files or default arduino sketch files
# set(${PROJECT_NAME}_SRCS)
set(${PROJECT_NAME}_HDRS User_Setup.h)

### Additional static libraries to include in the target.
# set(${PROJECT_NAME}_LIBS)

### Main sketch file
set(${PROJECT_NAME}_SKETCH untitled1.ino)

### Add project directories into the build
# add_subdirectory()

### Additional settings to add non-standard or your own Arduino libraries.
# For this example (libs will contain additional arduino libraries)
# An Arduino library my_lib will contain files in libs/my_lib/: my_lib.h, my_lib.cpp + any other cpp files
# link_directories(${CMAKE_CURRENT_SOURCE_DIR}/)

# For nested library sources replace ${LIB_NAME} with library name for each library
# set(_RECURSE true)

#### Additional settings for programmer. From programmers.txt
set(${PROJECT_NAME}_PROGRAMMER arduinoasisp)
set(${PROJECT_NAME}_PORT /dev/ttyUSB0)
# set(nano.upload.speed 9600)

## Verbose build process
set(${PROJECT_NAME}_AFLAGS -v)

generate_arduino_firmware(${PROJECT_NAME})
```

A search in the arduino-cmake project didn't give me anything useful, maybe someone here has an idea?

vsch, 2019-04-29:
@l0rn, what OS and CLion version are you running on?

I tried it on OS X with CLion 2018.3 and 2019.1 and both work as expected.

If you are using Windows, you need to follow Arduino CMake installation procedure (https://github.com/francoiscampbell/arduino-cmake/tree/dbd4fe372b9a0120325b429b3ed23c311ab3dc66#installing-clion--arduino-ide--mingw). On other systems you only need to have Arduino IDE installed in order for Arduino CMake to work.

l0rn, 2019-04-30:
Thanks fo your reply and sorry for forgetting this crucial information. I am using arch linux having the arduino toolchain installed (1.8.9-1)
CLion Version: 
```
CLion 2019.1.2
Build #CL-191.6707.69, built on April 18, 2019
JRE: 1.8.0_202-release-1483-b44 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Linux 4.19.34-1-MANJARO
```
Using Arduino Support 1.4.4 from 21.12.2018

fogx, 2020-01-23:
i have the same issue:
/snap/clion/100/bin/cmake/linux/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /home/bigpresenter/Private/Projects/Clion/radio
-- Generating radio
CMake Error at cmake/Platform/Arduino.cmake:860 (message):
  Invalid Arduino board ID (pro), aborting.

with:
```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 98)
set(PROJECT_NAME radio)

set(${PROJECT_NAME}_BOARD pro)
```

l0rn, 2020-03-05:
better late than never... arch linux seems to have a little deviance from standard.... this fixed it:
https://github.com/queezythegreat/arduino-cmake/issues/157#issuecomment-355550867

waterlubber, 2020-04-30:
I'm still having this problem. Symlinking the libraries does not help, nor does manually specifying boards.txt. I am on Archlinux as well with both arduino and the arch arduino SDK installed.