vsch/CLionArduinoPlugin #16: "Add to Targets" grayed out when making new classes

BoxWizard000000, 2019-07-22:
Just want to add separate classes to my project for organization. I'm new to CMake and half expected the plugin to be able to automatically append to the CMakeLists.txt so that new classes would be in the project.

Is there a workaround?

vsch, 2019-07-22:
@BoxWizard000000, support for refactoring and automatic modification of CMakeLists.txt is on the to do list.

You can add your class files to the lines commented with `### Source (.cpp) and Header (.h) files`

### Arduino CMakeLists.txt file

```
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD)

set(PROJECT_NAME)
set(${CMAKE_PROJECT_NAME}_BOARD)
set(ARDUINO_CPU)
project(${PROJECT_NAME})

### Source (.cpp) and Header (.h) files
set(${PROJECT_NAME}_SRCS)
set(${PROJECT_NAME}_HDRS)

### Additional static libraries to include in the target.
set(${CMAKE_PROJECT_NAME}_LIBS)

### Main sketch file
set(${CMAKE_PROJECT_NAME}_SKETCH)

### Add project sub-directories into the build
add_subdirectory()

### Additional settings to add non-standard or your own Arduino libraries.
link_directories()

# For nested library sources replace ${LIB_NAME} with library name for each library
set(${LIB_NAME}_RECURSE true)

#### Additional settings for programmer. From programmers.txt
set(${CMAKE_PROJECT_NAME}_PROGRAMMER)
set(${CMAKE_PROJECT_NAME}_PORT)
set(<$SET_BOARD$>.upload.speed)

## Verbose build process
set(${CMAKE_PROJECT_NAME}_AFLAGS -v)

### Sketch and Arduino Library projects
generate_arduino_firmware(${CMAKE_PROJECT_NAME})

### Static Library projects
generate_arduino_library(${CMAKE_PROJECT_NAME})
```

BoxWizard000000, 2019-07-22:
So I have a couple classes in a utils file. I've added them in like so:

```
set(${PROJECT_NAME}_SRCS utils/PIDFController.cpp)
set(${PROJECT_NAME}_HDRS User_Setup.h utils/PIDFController.hpp)
```

However in the files it still says "This file does not belong to any project". 

Is this normal?

closed 2019-07-22 by BoxWizard000000

vsch, 2019-08-22:
@BoxWizard000000, you probably need to reload CMakeLists.txt after it is modified so the IDE updates its configuration.