vsch/CLionArduinoPlugin #17: using set(${PROJECT_NAME}_SRCS ) has no effect on compile

rklosinski, 2019-08-22:
Just installed the latest version of this plugin. Skeleton code compiled and uploaded fine. So I tried adding a new class (test cpp/h) then added them to the CMakeList.txt file using set(${PROJECT_NAME}_SRCS test.cpp) and set(${PROJECT_NAME}_HDRS User_Setup.h test.h). This had no effect on the outcome of the compile. 

What did worked was to change set(${PROJECT_NAME}_SRCS) and set(${PROJECT_NAME}_HDRS User_Setup.h), then rename test.cpp to test.ino. It seems only ino files are searched for automatically, and set(${PROJECT_NAME}_SRCS) is broken.

vsch, 2019-08-22:
@rklosinski, did you reload the CMakeLists.txt after modifying it? The IDE needs to reload the file to change its project build files.

rklosinski, 2019-08-22:
yes I did

vsch, 2019-08-22:
@rklosinski, just figured out that if you want to include extra cpp files in the project then instead of `set(${PROJECT_NAME}_SKETCH untitled8.ino)` main file you should rename the `.ino` to `.cpp` and add all cpp file as `set(${PROJECT_NAME}_SRCS test.cpp untitled8.cpp)`

Loading CMakeLists.txt now recognizes all .cpp and .h files as part of the project. This must be somewhere in the docs of Arduino CMake but I don't remember coming across it.

With a sketch file all the .cpp files need to be included into the .ino file as part of its source.

rklosinski, 2019-08-26:
@vsch Thanks! I'll try that when Im able to get back to my Arduino project. another project's priority just bumped it down a notch.

jord1e, 2021-03-09:
For future reference (edit these two lines in your own project): https://github.com/francoiscampbell/arduino-cmake/pull/10

After this de _SKETCH, _SRCS and _HDRS all work.