vsch/CLionArduinoPlugin #18: FEATURE: Board NANO 33 BLE

xBitsSec, 2019-09-01:
It is possible to add the new Nano Boards?

[image]

Your CLion plugin is cool, THX for the hard work :)

vsch, 2019-09-01:
@xBitsSec, thank you for the compliment but I have not had any time to devote to this plugin since my initial work. I have not even had time to finish the Arduino project which prompted me to fork and update the plugin.  Having to earn a living certainly cramps my hobby projects. :(

1. If this board is supported by the new Arduino IDE then it should work because the plugin uses its board.txt config file.

    If there is a problem with not recognizing the new cpu although it is available in the Arduino IDE then please let me know. It is possible that the format for the file has changed and I need to update the plugin parsing of this file.

2. If there is no support yet and you know how to create the definitions for this cpu in boards.txt then you can create your own copy of the file with updates for the new CPU.

Other that these two options I do not see what else can be done at the present time.

Nailer, 2019-12-12:
I have been unable to get the BLE33 to work as well, as it uses a different chipset than the other supported Arduinos. I Tried fiddling with custom boards.txt files, but could not get it to work.