vsch/CLionArduinoPlugin #20: Need help setting up the toolchain

RoscoeTheDog, 2019-11-17:
I read the Read Me. Also lurked around for awhile trying to figure out the toolchain. This looks like a great pluggin and I would love to use clion for development instead of the arduino IDE.

Upon installating the pluggin through CLion and restarting the IDE, I can then create a new project from the 'Arduino Sketch' menu. I select C++ 17, my board (Arduino Uno), the port it's connected on (COM 3), a baud rate and hit 'create'.

I'm then greeted with a new project with an auto-generated cmakelist file such as this;
```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 17)
set(<@PROJECT_NAME@> untitled7)
set(${<@PROJECT_NAME@>}_BOARD uno)              // <> is higlighted
# set(ARDUINO_CPU)
project(${<@PROJECT_NAME@>})                    // <> is higlighted
# set(${<@PROJECT_NAME@>}_SRCS)
# set(${<@PROJECT_NAME@>}_HDRS)
# set(${<@PROJECT_NAME@>}_LIBS)
# link_directories(${CMAKE_CURRENT_SOURCE_DIR}/)
# add_subdirectory()
# set(${<@PROJECT_NAME@>}_SKETCH untitled7.ino)
set(${<@PROJECT_NAME@>}_PROGRAMMER avrispmkii)  // <> is higlighted
set(${<@PROJECT_NAME@>}_PORT COM3)              // <> is higlighted
# set(${<@PROJECT_NAME@>}_AFLAGS -v)
set(uno.upload.speed 256000)
generate_arduino_firmware(${PROJECT_NAME})
```

Clion immediately detects some syntax issues with this Cmake file however. I am a little unfamiliar with Cmake still unfortunately, so I don't know if this is a false flag or if I need to change something. I will put comments on the lines they are underlined.

Besides the fact of this minor issue, I cannot seem to figure out how to create a new ino sketch. This should be a fairly easy task but it still eludes me. I have tried right clicking the project directory folder, new -> "Arduino Sketch". This then prompts me to enter a Sketch Name. I enter in something and hit Ok but nothing visually apparent happens. I'm not sure if this is a bug or I'm just not understanding this clearly. Shouldn't it create a new INO file with a basic loop?

It might be worth mentioning that I did not install these other two plugins available from the Clion plugin Repo: "Arduino" by Francois Campbell plugin or the "Serial Port Monitor" by Dmitry Cherkas.

I have only installed this forked version "Arduino Support" as it was filed under the category of "Tools and Integration" and seemed very polished.

vsch, 2019-11-17:
@RoscoeTheDog, when a new project is created an empty sketch file is already part of it: see the line with `set(${<@PROJECT_NAME@>}_SKETCH untitled7.ino)`, the sketch name is untitled7.ino

CLion project view is a bit counter intuitive compared to other JetBrains IDEs, and the arduino cmake includes dependencies which show up as files in the project but outside the project folder. This adds to initial confusion.

[image]

RoscoeTheDog, 2019-11-17:
Thank you for the speedy reply. I see now where the sketch name is defined in Cmake but I do not have any of the dependencies or the sketch file generated in the root directory. Is there any additional steps we are supposed to take to setting up the enviroment? I figured this would be done automatically upon creating a new project.

Here is what mine looks like upon creating a new project.
[image]

I also do have the basic Arduino IDE installed too, so the libraries and dependencies should be installed on my system.

vsch, 2019-11-17:
@RoscoeTheDog, it looks like there is no sketch. So something is fishy. What version of CLion are you using?

Your CMakeLists.txt is wrong. It is the raw template that is used for processing. Not the processed result:

```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 17)
set(PROJECT_NAME untitled9)

set(${PROJECT_NAME}_BOARD uno)
# set(ARDUINO_CPU)
project(${PROJECT_NAME})

# Define additional source and header files or default arduino sketch files
# set(${PROJECT_NAME}_SRCS)
# set(${PROJECT_NAME}_HDRS)

### Additional static libraries to include in the target.
# set(${PROJECT_NAME}_LIBS)

### Main sketch file
set(${PROJECT_NAME}_SKETCH untitled9.ino)

### Add project directories into the build
# add_subdirectory()

### Additional settings to add non-standard or your own Arduino libraries.
# For this example (libs will contain additional arduino libraries)
# An Arduino library my_lib will contain files in libs/my_lib/: my_lib.h, my_lib.cpp + any other cpp files
# link_directories(${CMAKE_CURRENT_SOURCE_DIR}/..)

# For nested library sources replace ${LIB_NAME} with library name for each library
# set(_RECURSE true)

#### Additional settings for programmer. From programmers.txt
set(${PROJECT_NAME}_PROGRAMMER avrispmkii)
set(${PROJECT_NAME}_PORT /dev/cu.usbserial-00000000)
# set(uno.upload.speed 9600)

## Verbose build process
set(${PROJECT_NAME}_AFLAGS -v)

generate_arduino_firmware(${PROJECT_NAME})
```

BTW, you do not need the other two plugins. "Arduino" is the project I forked to make fixes and improvements. So this plugin supersedes it. 

I also folded the functionality of the serial port monitor into this plugin to disconnect from the port when the project is built and re-connect after the build is finished. Otherwise upload, if part of the build, will fail because the port is already used.

[image]

Unfortunately, I am absolutely swamped with other commitments and do not have time to spend on this project but I have not abandoned. I usually reply

vsch, 2019-11-17:
@RoscoeTheDog are you getting exceptions in the IDE? If you get one there is a little flashing red circle in the bottom right corner of the status bar.

vsch, 2019-11-17:
@RoscoeTheDog, are you on Windows? If you are there is a whole procedure to follow for installation of the tools needed. I have not done it since I am on OS X and running Windows on parallels makes the experience 10x worse. There are a few discussions covering installation of arduino tools for Arduino CMake on Windows in other issues in this repo.

RoscoeTheDog, 2019-11-17:
It should be the latest. I just double checked from JetBrains Toolbox.

CLion 2019.2.5
Build #CL-192.7142.39, built on October 29, 2019
Licensed to Aspen Mallery
Subscription is active until October 19, 2020
For educational use only.
Runtime version: 11.0.4+10-b304.77 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 7 6.1
GC: ParNew, ConcurrentMarkSweep
Memory: 1987M
Cores: 8
Registry: run.processes.with.pty=TRUE
Non-Bundled Plugins: com.markskelton.one-dark-theme, com.samdark.intellij-visual-studio-code-dark-plus, com.vladsch.clionarduinoplugin, mobi.hsz.idea.gitignore

I will try installing this on my W10 laptop and see if I can reproduce the problem. I'm also very curious in what could be causing this issue.

The only installed plugins are:
Arduino Support (this current one)
.ignore (for github)
One Dark Theme.

vsch, 2019-11-17:
@RoscoeTheDog, you are using windows. So the issue is most likely related to following the installation instructions for Windows for Arduino CMake.

The discussion at the bottom of #9 may be helpful.

RoscoeTheDog, 2019-11-17:
I wanted to add here that the problem seems to be reproducing itself on W10 x64 as well.

Unfortunately, following all these instructions from the link you mentioned pretty much defeats the purpose of using Clion. Kind of a bummer as I've enjoyed working with cpp projects in it.

The older version  _"Arduino" by Francois Campbell plugin_ actually does generate projects correctly however. This confirms that it is not solely a matter of missing the modules/libraries and other core resources.

I understand very well what it is like to be busy with too many commitments and projects. I hope you and any other devs can work out the Windows support issues for this plugin. I'll have to revisit it again sometime in the future and see if it works out for me then.

vsch, 2019-11-17:
@RoscoeTheDog, I will mark this issue as a bug until I can validate it or fix it.