vsch/CLionArduinoPlugin #22: Open Existing Project files

Syzygianinfern0, 2020-01-11:
How do I set up the plugin to work with files and folders already present? I just get a `CMake project is not loaded` and a `Select CMakeLists.txt file...` error on top.

vsch, 2020-01-11:
@Syzygianinfern0, the plugin does not do existing project import. That would require more knowledge of CMAKE and CLion that I have.

CLion works strictly with CMakeLists.txt and the plugin creates this but does not fully integrate with CLion to modify and update the CMakeLists.txt file as CLion does with its native projects.

The only workaround I can see is to: 
1. create a new project
1. copy existing project files to it
1. manually edit CMakeLists.tst to add the existing project files

Syzygianinfern0, 2020-01-12:
How do I upload files other than main .cpp file to the Arduino?

vsch, 2020-01-12:
If I understand you correctly, you have a project which consists of multiple source files. 

You need to add all the source and header files to the CMakeLists.txt so that they are built into a single image which will be loaded to the Arduino.

Your question of uploading other files than main .cpp implies that your project is not a single executable but multiple sketches in a single project.

I don't know how this should be handled. I suspect you need to split them up into separate projects, each one producing a single executable or library.

Syzygianinfern0, 2020-01-12:
> You need to add all the source and header files to the CMakeLists.txt so that they are built into a single image which will be loaded to the Arduino.

Can I use "\*.cpp" and "\*.h" type of method to add them?

Syzygianinfern0, 2020-01-12:
This is my directory structure https://hastebin.com/udeguduzom.coffeescript
This is my CMakeLists.txt  https://hastebin.com/zosagetibo.shell

What is missing?
Error log https://hastebin.com/noyigugobe.rb

vsch, 2020-01-12:
I never even considered using wildcards for specifying source and header files. I doubt that  [Arduino CMake](https://github.com/francoiscampbell/arduino-cmake) can handle it. I would list each file explicitly.

Syzygianinfern0, 2020-01-12:
Can you please go through the above listed links and assure that I'm not doing any blunder

vsch, 2020-01-12:
@Syzygianinfern0, no I cannot. This is your learning process and I cannot do it for you.