vsch/CLionArduinoPlugin #23: Add a custom baudrate

flolbr, 2020-01-30:
Hi, I'm facing the need to use a custom baudrate that is not listed in the settings' list.
Is there any option to add a custom baudrate in the list ? If not, would you please consider adding this possibility ?

Many thanks

vsch, 2020-01-30:
@flolbr, I have not tried it but it might work. Have you tried editing the baud rate in the CMakeLists.txt file to your baud rate?

It will not affect the serial monitor tool window but should work for upload to the board.

I will add this as an enhancement to allow custom baud rates in the project settings and serial monitor tool window.

oxygene, 2020-02-08:
I wanted to migrate my project from Arduino IDE to CLion and I figured that my sketch was using a baud rate of 1_000_000 which is not available in the clion plugin right now. So implementing this enhancement would be great added value!