vsch/CLionArduinoPlugin #28: custom programmers.txt not used

GeorgeGkinis, 2020-02-18:
Hello!

Thanks for our work.

I am trying to use a generic nRF5 board with Arduino in Clion.
I have successfully selected the generic nRF5 board but i cannot seem to be able to select a proper programmer :
[image]
[image]

When creating a new project, only the bundled ones are shown in the drop-down list.
[image]

I could manually add the programmers to the bundled ones, but cannot seem to be able to find the bundled programmers.txt fle.

Any ideas on how to resolve?

vsch, 2020-02-18:
@GeorgeGkinis, the bundled programmers and boards are part of Arduino development installation. On OS X it is in `~/Library/Arduino15`, ie library directory under user home directory. If memory serves correctly it would be in  `~/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5` directory.

This is the reason they are copied to user directory where they can be modified when bundled checkbox is disabled to allow modifying ones used by the plugin without modifying ones used by Arduino IDE.

GeorgeGkinis, 2020-02-19:
Thanks! I will try to add the programmers to the default "bundled" programmers.txt file.

By the way, I did manually specify the programmers.txt file to be used for my board but it had no effect!
I still saw the bundled ones.
Check the first screenshot.
Bug maybe?

GeorgeGkinis, 2020-02-20:
After messing around a bit i found out that the bundled programmers.txt is indeed bundled.
I did : 

```
vagrant@linux:/snap/clion/current/lib$ locate programmers.txt
/home/vagrant/.arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/programmers.txt
/home/vagrant/.platformio/packages/framework-arduinoespressif32/programmers.txt
/home/vagrant/nRF5/arduino-1.8.12/hardware/arduino/avr/programmers.txt
```

**/home/vagrant/.arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/programmers.txt** is the one I want to use.
**/home/vagrant/nRF5/arduino-1.8.12/hardware/arduino/avr/programmers.txt** is the one that came with my arduino installation.

Even if I add my new programmers to the second one the drop down list does not change.
Even if I make the second one empty, I still get the same in the drop down list : 
```
vagrant@linux:/snap/clion/current/lib$ cat /home/vagrant/nRF5/arduino-1.8.12/hardware/arduino/avr/programmers.txt

vagrant@linux:/snap/clion/current/lib$ 
```

It seems that the "bundled" **programmers.txt** is somehow hard coded within the plugin and does not reside on the file system.

I still cannot select a custom **programmers.txt**. It is simply ignored.

I see that [here](https://github.com/vsch/CLionArduinoPlugin/blob/3c1e7875649dba2a3ad83ee0edb82febcaa00fcc/src/com/vladsch/clionarduinoplugin/resources/ArduinoConfig.kt#L73) the programmer.txt file is bundled as a resource within the jar.

GeorgeGkinis, 2020-03-06:
Hello!

Any updates concerning my last comment?

Is it possible to use a custom programmers.txt file without the setting being ignored?

Nailer, 2020-04-03:
I'd really like to get this plugin to work with Nano 33 IoT. Eagerly awaiting an official update on supporting the new cards :)

Wimmie, 2021-01-13:
> 
> I see that [here](https://github.com/vsch/CLionArduinoPlugin/blob/3c1e7875649dba2a3ad83ee0edb82febcaa00fcc/src/com/vladsch/clionarduinoplugin/resources/ArduinoConfig.kt#L73) the programmer.txt file is bundled as a resource within the jar.
> 
> seems to be a bug after all.

There is a whacky way to fix this. You should go to the plugins dir of CLion, see https://www.jetbrains.com/help/idea/tuning-the-ide.html#default-dirs. There you will find the plugin (_CLionArduinoPlugin.zip_), extract the _CLionArduinoPlugin.jar_. You can exctract that one as well, or open it like I did with WinRAR. Then find _com\vladsch\clionarduinoplugin\config\programmers.txt_. Open that file in a text editor, add the programmers, put it back in the jar and put the jar back into the CLionArduinoPlugin.zip.