vsch/CLionArduinoPlugin #35: Feature request: clear all button on serail monitor.

raphael-bmec-co, 2020-04-28:
Hello

Great tool.

A small feature request:

Clear all is available from right click menu in serial monitor. It would be great to have trash can icon on the left hand side.

vsch, 2023-03-16:
@raphael-bmec-co, actually it used to have one, as provided by the console implementation of  the IDE, but the implementation class has changed in the IDE as of 2020.1, I think, and the button disappeared.

I am fixing it for next release. I also added an option for automatic clearing of the console on reconnect after build (or successful build as selected in the options). That way the serial console is not cluttered with old runs.

vsch, 2023-03-16:
I will not be making a release until I can get CMake fixed for 2022.3.

In the mean time I have a build I made for this fix for me to use. You can download it and install plugin from disk in the IDE: https://github.com/vsch/CLionArduinoPlugin/blob/master/dist/CLionArduinoPlugin.1.5.7.zip 

Compatibility is up to 2022.2.

closed 2023-03-16 by vsch