vsch/CLionArduinoPlugin #36: Compiler error when building the simplest of code

jbasiago, 2020-04-28:
I just installed CLion 2020.1 and this plugin.  Created a sketch.  Added some simple code to generated ino file.  I've assured that the arduino SDK was loaded.  Working on a Mac.  I'm getting this error on build.

====================[ Build | upload | Debug ]==================================
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake --build /Users/jbasiago/Development/Arduino/AutomaticUncoupler/cmake-build-debug --target upload -- -j 4
[  3%] Building CXX object CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp.obj
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp:29:46: error: expected initializer before 'noexcept'
 void * operator new(size_t size, void * ptr) noexcept {
                                              ^~~~~~~~
make[3]: *** [CMakeFiles/uno_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp.obj] Error 1
make[2]: *** [CMakeFiles/uno_CORE.dir/all] Error 2
make[1]: *** [CMakeFiles/upload.dir/rule] Error 2
make: *** [upload] Error 2

I have no idea how to get past this.
Thanks.

jbasiago, 2020-04-28:
Arduino Version 1.8.12

KilianSteenman, 2020-05-02:
@jbasiago Make sure you are using a language standard higher than the default (c++ 98). Using C++ 17 fixed this issue for me.

jbasiago, 2020-05-02:
That did it.  Thanks.

closed 2020-05-02 by jbasiago