vsch/CLionArduinoPlugin #40: link_directories() does not link Arduino library

quarter28, 2020-06-07:
Providing path to Arduino library tree:

my_lib/
  |--my_lib.cpp
  `--my-lib.h

using

link_directories(<path to my_lib>)
 or
link_directories(${CMAKE_CURRENT_SOURCE_DIR}/<relative path to my_lib>)

Does nothing, apparently.

RoscoeTheDog, 2020-10-19:
Any updates on this? I also had the same issue. 

Even the suggestions [from stackoverflow](https://stackoverflow.com/questions/48361713/clion-arduino-undefined-referencel) did not work.

I actually got header files to link correctly. But it seems the sourcees to those headers cannot still.

This issue, along with the fact that I am still having issues uploading to the board, makes this plugin un-unusable to me ATM :(

fabianeichfeldt, 2020-12-18:
I could solve my issue with this stackoverflow thread: https://stackoverflow.com/questions/48361713/clion-arduino-undefined-reference
important things: 
 - rename *.ino to *.cpp
 - add glob with headers and cpp files and main cpp file

```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(CMAKE_CXX_STANDARD 98)
set(PROJECT_NAME FlightControl)

set(${PROJECT_NAME}_BOARD megaADK)
# set(ARDUINO_CPU)
project(${PROJECT_NAME})

# Define additional source and header files or default arduino sketch files
include_directories(${PROJECT_SOURCE_DIR})
file(GLOB SRC_FILES ${PROJECT_SOURCE_DIR}/src/*.cpp)
file(GLOB HDR_FILES ${PROJECT_SOURCE_DIR}/src/*.h)

set(PROJ_SRC ${SRC_FILES} <myproject>.cpp)
set(PROJ_HDR ${HDR_FILES})

generate_arduino_firmware(${CMAKE_PROJECT_NAME}
        SRCS ${PROJ_SRC}
        HDRS ${PROJ_HDR}
        )
```

Hope this helps anyone