vsch/CLionArduinoPlugin #41: Error when trying to make and upload

morasz, 2020-06-09:
Hi there!

I wanted to use your plugin with an Arduino Nano (Old Bootloader) but encountered an issue when I hit the "build" button. (Same issue is described here: https://www.gitmemory.com/slawalata)

`====================[ Build | upload | Debug ]==================================
/Applications/CLion.app/Contents/bin/cmake/mac/bin/cmake --build /Users/silasborowy/CLionProjects/untitled2/cmake-build-debug --target upload -- -j 6
[  3%] Building CXX object CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp.obj
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp:29:46: error: expected initializer before 'noexcept'
 void * operator new(size_t size, void * ptr) noexcept {
                                              ^~~~~~~~
make[3]: *** [CMakeFiles/nano_CORE.dir/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/new.cpp.obj] Error 1
make[2]: *** [CMakeFiles/nano_CORE.dir/all] Error 2
make[1]: *** [CMakeFiles/upload.dir/rule] Error 2
make: *** [upload] Error 2`

Is that my fault? I use:
CLion 2020.1 #CL-201.7846.88
Intellij Arduino Support plugin v1.5.2
MacOS 10.15.5

Best,
Silas

adth03, 2020-06-20:
I had the same issue. Using "language standard" C++ 17 should work.

ahoeh, 2021-01-01:
Hey i run into same problem...
Please try chaning CMAKE_CXX_STANDARD parameter and report if that solved it.

Following solved my problem:
* in CMakeLists.txt: `set(CMAKE_CXX_STANDARD 11)`

Good luck. Best regards
ahoeh