vsch/CLionArduinoPlugin #52: Programming Arduino Leonardo (ATmega32U4)

melvin2204, 2021-04-02:
I'm facing problems with programming my Arduino Leonardo. It works fine from the Arduino IDE, but in CLion it gets stuck at `Connecting to programmer: .`.

The Arduino IDE seems to connect to the board with a baudrate of 1200bps and then immediately disconnecting. This causes the board to be reset and open a new serial port. The IDE then uploads the code via that serial port. See this SO answer: https://stackoverflow.com/a/46818616

For some reason the CLion Arduino plugin doesn't do this. I can do this manually by briefly connecting the RST pin to GND and checking the new COM port. Often it is just the old COM port + 1, so if normally it has COM3, then after resetting it will have COM4. Then I change `set(${PROJECT_NAME}_PORT COM3)` to `set(${PROJECT_NAME}_PORT COM4)` in the CMakeLists.txt file. To upload the code, I manually reset the Arduino by briefly shorting RST and GND and immediately hitting upload. Then it successfully uploads the code.

It is quite a chore to do this every time I want to upload something. Is it possible that the plugin does this automatically, just like the Arduino IDE does by briefly connecting with 1200 bps?

aimass, 2022-04-15:
I'm having the same issue with a Gun using the USB port (CLion + Arduino Plugin does not support WiFi-based upload yet).

I found a possible easy fix for Unix-like systems (macOS, Linux, *BSD) this bit of shell script right before avrdude should do the trick (I have seen similar tricks on Windows using similar methods):
https://stackoverflow.com/a/46040407/2397947

What I'm not too clear is where to stick this in the cmake file :-(

This should be an easy fix for @vsch or in the mean time, tell us where we can stick the 1200 bps reset thingy in the CMake toolchain. 

I'll follow up if I get it working on my own.

Best,

-- 
Alex

melvin2204, 2022-04-17:
Hi Alex,

Since this project has been dead for over 2 years now, I don't think any bug fixes will be done. 

Personally, I have now switched to Platform.io, and never looked back at this plugin since. Platform.io supports over 1000 different boards and 27 frameworks. It is also compatible with many more IDEs. I don't want this to sound like an advertisement, but it made my development so much easier. It is free and open source. You should really consider trying it: https://platformio.org/

With kind regards,

Melvin

aimass, 2022-04-18:
Hi Melvin, 

Thanks for your prompt reply and comments ! platform.io is is in my evaluation list so I’ll be sure to give it a test drive. My code has gotten to a point where I’m seriously thinking of investing in an ICE and be able to debug in-circuit. But the Microchip software is Windows-only so I’ve been looking for alternatives to more professionally develop with the Atmel 8bit AVRs 

Thanks again!
Alex