vsch/CLionArduinoPlugin #56: Keep getting IDE Fatal Errors with CLion 2021.1.1

EasyG0ing1, 2021-05-30:
I constantly get IDE errors in Clion version 2021.1.1 I believe they happen when I close a project...

Here is the stack trace:

```
com.vladsch.clionarduinoplugin.serial.SerialProjectComponent@65941d3f

com.intellij.serviceContainer.AlreadyDisposedException: Already disposed: Project(name=BlockNot, containerState=DISPOSE_IN_PROGRESS, componentStore=/Users/michael/CLionProjects/BlockNot) (disposed)
	at com.intellij.serviceContainer.ComponentManagerImpl.getMessageBus(ComponentManagerImpl.kt:211)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.fireStateChanged(ToolWindowManagerImpl.kt:1640)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.unregisterToolWindow(ToolWindowManagerImpl.kt:1107)
	at com.jetbrains.rdserver.toolWindow.BackendServerToolWindowManager.unregisterToolWindow(BackendServerToolWindowManager.kt:70)
	at com.vladsch.clionarduinoplugin.serial.SerialMonitorToolWindow.unregisterToolWindow(SerialMonitorToolWindow.java:72)
	at com.vladsch.clionarduinoplugin.serial.SerialProjectComponent.projectClosed(SerialProjectComponent.java:87)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.fireProjectClosed(ProjectManagerImpl.java:467)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.lambda$closeProject$4(ProjectManagerImpl.java:376)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:959)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.closeProject(ProjectManagerImpl.java:365)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.closeAndDispose(ProjectManagerImpl.java:400)
	at com.intellij.openapi.wm.impl.CloseProjectWindowHelper.closeProjectAndShowWelcomeFrameIfNoProjectOpened(CloseProjectWindowHelper.kt:52)
	at com.intellij.openapi.wm.impl.CloseProjectWindowHelper.windowClosing(CloseProjectWindowHelper.kt:40)
	at com.intellij.openapi.wm.impl.ProjectFrameHelper$2.windowClosing(ProjectFrameHelper.java:228)
	at java.desktop/java.awt.AWTEventMulticaster.windowClosing(AWTEventMulticaster.java:357)
	at java.desktop/java.awt.AWTEventMulticaster.windowClosing(AWTEventMulticaster.java:357)
	at java.desktop/java.awt.Window.processWindowEvent(Window.java:2090)
	at java.desktop/javax.swing.JFrame.processWindowEvent(JFrame.java:298)
	at java.desktop/java.awt.Window.processEvent(Window.java:2049)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5027)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2784)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4859)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:778)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:751)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:749)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:748)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:969)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:839)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:449)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:808)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$9(IdeEventQueue.java:448)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:781)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:502)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

bbar2, 2021-07-09:
I'm surprised at how little response your issue has raised.  I've experienced the same thing.  When CLion went to 2021.1, this plugin broke.  Maybe very few people use this plugin for Arduino development using CLion, and it will no longer be supported.  This plugin probably falls into the category of something fun to create, but a pain to maintain.  I am still surprised that no other users are having these problems.  It was nice while it lasted.  I've noted other plugins commenting on CLion's lack of response to plugin issues created by the 2021 update.   I've switched to VSCode with the PlatformIO extension.  Even though I prefer CLion's environment, I like the fact that the VSCode/PlatformIO approach does a better job of finding the right serial port.  If you ever find a work around, I hope you post the results here.

EasyG0ing1, 2021-07-11:
@bbar2 - When I develop in CLion for Arduino, I USUALLY have the Arduino IDE open as well with a sketch loaded that I simply call "delete.ino" ... CLion gives me code completion like nobody else can ... lol ... then when I want to test my code, I simple CMD+A, CMD+C CMD+TAB to Arduino IDE, CMD+V then CMD+U ... takes less than 3 seconds when I'm dialed in. It would be nice to use Clion for all of it, but I've had too many random issues with compiling and uploading when I switch to a different MC like an ESP8266 or a Teensy ... or programming an ATTIny85 through a nano ... I wouldn't even attempt that with PlatformIO...  the Raspberry Pi Pico has been nice to develop on with CLion since the code gets uploaded to a volume instead of the serial port ...

And as far as this plugin goes, It's been removed ... it doesn't like the new version of CLion and I got no time for plugins that won't play ball.  lol

Mike

bbar2, 2021-07-13:
Thanks for the work around.  Sometimes brute force is the right answer.  And like you said -- it's a matter of seconds. 
VSCode's code completion and class navigation is not as nice as CLion's, but it's not bad and the rest of the environment is certainly working out to be more user friendly.  No CMake is nice, and port selection also works well even with different projects on two targets from the same dev machine.

jskubick, 2021-12-16:
**Is there any version of CLion known to actually _work_ with v1.52 of this plugin?** Last night, I spent about 2 hours installing and uninstalling older and older versions of CLion trying to find one that actually works without blowing up in a manner similar to above, but all of the following versions died on me:
* 2020.3.4
* 2020.1.3
* 2019.3.6
* 2019.2.5

The text of OP's exceptions (similar to what I remember seeing) leads me to think it might be blowing up due to some security constraint newly imposed by Jetbrains' JRE that was retroactively applied to later releases of 2020.3, 2020.1, 2019.3, 2019.2, and 2019.1. In other words, a big architectural change (CMake?) in 2021.1+ finished the job, but its PRESENT problem with newer releases of older versions might be JRE-related.

That suggests two things that, when done together, might be able to make the existing Arduino plugin work:
* Find the last release of CLion that's known (or at least believed to be) architecturally compatible with this plugin
* Work backwards from the penultimate release of that last known-good version of CLion until one with a JRE that won't blow up with a security exception is found. This assumes Jetbrains regards changes to the JRE as a minor-version incrementing change. If they went back and updated the JREs of ALL archived releases, the strategy obviously wouldn't work.

Or... possibly... as an alternative to working backwards to find a compatible major release with equally compatible JRE, try to figure out whether there's some way to make the newer JRE _not_ blow up with a security exception (kind of like how strong encryption used to stupidly be disabled by default unless you jumped through hoops to create a security exception policy for the machine).

bbar2, 2021-12-16:
The plugin started working again with the CLion 2021.2 release.

jskubick, 2021-12-16:
Hmmm, I just reinstalled 2021.3, vCL-213.5744.254, and it still appears to be broken. What specific version of CLion are you presently running?

I created a new Arduino sketch after installing the plugin and restarting, and got two fatal errors:

~~~
com.vladsch.clionarduinoplugin.serial.SerialProjectComponent@58dc1b60

com.intellij.serviceContainer.AlreadyDisposedException: Already disposed: Project(name=vsaMega1, containerState=DISPOSE_IN_PROGRESS, componentStore=C:\src\experiments\CLion\vsaMega1) (disposed)
	at com.intellij.serviceContainer.ComponentManagerImpl.getMessageBus(ComponentManagerImpl.kt:191)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.fireStateChanged(ToolWindowManagerImpl.kt:1712)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.unregisterToolWindow(ToolWindowManagerImpl.kt:1125)
	at com.jetbrains.rdserver.toolWindow.BackendServerToolWindowManager.unregisterToolWindow(BackendServerToolWindowManager.kt:108)
	at com.vladsch.clionarduinoplugin.serial.SerialMonitorToolWindow.unregisterToolWindow(SerialMonitorToolWindow.java:72)
	at com.vladsch.clionarduinoplugin.serial.SerialProjectComponent.projectClosed(SerialProjectComponent.java:87)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.fireProjectClosed(ProjectManagerImpl.java:479)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.lambda$closeProject$4(ProjectManagerImpl.java:384)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:947)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.lambda$closeProject$5(ProjectManagerImpl.java:373)
	at com.intellij.openapi.util.ShutDownTracker.executeWithStopperThread(ShutDownTracker.java:114)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.closeProject(ProjectManagerImpl.java:354)
	at com.intellij.openapi.project.impl.ProjectManagerImpl.closeAndDispose(ProjectManagerImpl.java:406)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl$closeAndDisposeKeepingFrame$1.compute(ProjectManagerExImpl.kt:364)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl$closeAndDisposeKeepingFrame$1.compute(ProjectManagerExImpl.kt:59)
	at com.intellij.openapi.wm.impl.WindowManagerImpl.runWithFrameReuseEnabled(WindowManagerImpl.kt:379)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl.closeAndDisposeKeepingFrame(ProjectManagerExImpl.kt:364)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl.access$closeAndDisposeKeepingFrame(ProjectManagerExImpl.kt:59)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl$checkExistingProjectOnOpen$1.invoke(ProjectManagerExImpl.kt:346)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl$checkExistingProjectOnOpen$1.run(ProjectManagerExImpl.kt:59)
	at com.intellij.openapi.application.impl.ApplicationImpl.invokeAndWait(ApplicationImpl.java:437)
	at com.intellij.openapi.application.impl.ApplicationImpl.invokeAndWait(ApplicationImpl.java:455)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl.checkExistingProjectOnOpen(ProjectManagerExImpl.kt:310)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl.openProjectAsync(ProjectManagerExImpl.kt:103)
	at com.intellij.openapi.project.impl.ProjectManagerExImpl.openProject(ProjectManagerExImpl.kt:75)
	at com.intellij.ide.util.projectWizard.AbstractNewProjectStep.doGenerateProject(AbstractNewProjectStep.java:243)
	at com.intellij.ide.util.projectWizard.AbstractNewProjectStep$AbstractCallback.consume(AbstractNewProjectStep.java:188)
	at com.intellij.ide.util.projectWizard.ProjectSettingsStepBase$1.actionPerformed(ProjectSettingsStepBase.java:143)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:270)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6654)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3345)
	at java.desktop/java.awt.Component.processEvent(Component.java:6419)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5029)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4861)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2790)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4861)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:778)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
~~~

bbar2, 2021-12-16:
I'm using CLion 2021.2 on a Mac.  
Build #CL-212.4746.93, built on July 27, 2021
Runtime version: 11.0.11+9-b1504.13 x86_64
macOS 12.1

Although I confirmed the plugin worked when 2021.2 came out, I used it only briefly.  When the plugin was totally dead, prior to 2021.2, I switched to PlatformIO and have not looked back.  No more of these problems.

jskubick, 2021-12-16:
Well, for what it's worth, I uninstalled 2021.3 and tried 2021.2.4... and it's still as dead as John Cleese's parrot:

~~~
java.lang.NullPointerException
	at com.vladsch.clionarduinoplugin.resources.ArduinoToolchainFiles$1.run(ArduinoToolchainFiles.java:55)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:936)
	at com.vladsch.clionarduinoplugin.resources.ArduinoToolchainFiles.copyToDirectory(ArduinoToolchainFiles.java:25)
	at com.vladsch.clionarduinoplugin.generators.ArduinoProjectGenerator.createFiles(ArduinoProjectGenerator.kt:153)
	at com.vladsch.clionarduinoplugin.generators.ArduinoProjectGenerator.access$createFiles(ArduinoProjectGenerator.kt:59)
	at com.vladsch.clionarduinoplugin.generators.ArduinoProjectGenerator$generateProject$1.compute(ArduinoProjectGenerator.kt:167)
	at com.vladsch.clionarduinoplugin.generators.ArduinoProjectGenerator$generateProject$1.compute(ArduinoProjectGenerator.kt:59)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteActionWithClass(ApplicationImpl.java:924)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:950)
	at com.vladsch.clionarduinoplugin.generators.ArduinoProjectGenerator.generateProject(ArduinoProjectGenerator.kt:166)
	at com.vladsch.clionarduinoplugin.generators.ArduinoProjectGenerator.generateProject(ArduinoProjectGenerator.kt:59)
	at com.intellij.ide.util.projectWizard.AbstractNewProjectStep.doGenerateProject(AbstractNewProjectStep.java:231)
	at com.intellij.ide.util.projectWizard.AbstractNewProjectStep$AbstractCallback.consume(AbstractNewProjectStep.java:180)
	at com.intellij.ide.util.projectWizard.ProjectSettingsStepBase$1.actionPerformed(ProjectSettingsStepBase.java:141)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:270)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6652)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3345)
	at java.desktop/java.awt.Component.processEvent(Component.java:6417)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5027)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4859)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2784)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4859)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:778)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:751)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:749)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:748)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:885)
	at com.intellij.ide.IdeEventQueue.dispatchMouseEvent(IdeEventQueue.java:814)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:751)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:441)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:825)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:440)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:794)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:492)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
~~~

bbar2, 2021-12-16:
that's a bummer.  When this first happened to me on the 2021.1 update, I was unsuccessful rolling back to the prior version where the plugin had been working with no issues.  So I don't have complete confidence in the rollback process.  This is a frustrating problem.  You know it's something simple, but with the plugin basically being unsupported, the problem lingers.

jskubick, 2021-12-17:
OMG, CLion's configuration is like a fsck'ing virus. It's like they want to make it as hard as they possibly can to scrub the slate clean and start over with a completely 100% virgin installation, untainted by past settings and plugins. 

I've uninstalled it & told Windows to delete all the settings and files.

I then went and manually deleted the files under c:\Program Files\Jetbrains\CLion* that it left behind, and did the same with the CLion* directories under AppData\Local\Jetbrains and AppData\Roaming\Jetbrains.

I then reinstalled Clion 2021.3... and it all came back like a bad horror movie... plugins, settings, *everything* :-O

vsch, 2023-03-11:
I uploaded a new release 1.5.6 which works on CLion 2019.1 to 2022.2. 

It seems CLion 2022.3 has made big changes to their CMake and arduino-cmake does not work. I have zero experience with CMake, I have a perpetual fallback license for CLion 2020.2 and I am not using sufficiently to justify making it work for later versions.

EasyG0ing1, 2023-03-21:
@vsch 

>It seems CLion 2022.3 has made big changes to their CMake and arduino-cmake does not work. I have zero experience with CMake, I have a perpetual fallback license for CLion 2020.2 and I am not using sufficiently to justify making it work for later versions.

If I'm not mistaken, you can get free full licensing of Jetbrains IDEs as long as you can demonstrate that you are an active open source developer ... [Here is the info](https://sales.jetbrains.com/hc/en-gb/articles/360016581839-What-is-an-OSS-development-license-and-who-qualifies-for-one-) on it.

vsch, 2023-03-22:
@EasyG0ing1, that is only part of the issue. Maintaining plugins for the IDE is a real PITA. The API keeps evolving with every release, including preview releases. This means constant effort to keep the plugin compatible with the new IDEs.

Additionally, CLion has obfuscated a large chunk of its implementation starting in version 2020.3 (if memory serves correctly). This means that the only way I found of compiling the plugin is to copy the `clion.jar` from an older version to the project directory. The plugin runs fine with newer IDEs for now, but I have to actually run it to find out if it works. No compile time checking, since I have to build against an older IDE implementation.

The JetBrains Open API is a jungle of massive proportions, constantly evolving, and a nightmare to figure out, unless you spend a lot of time doing it because of all the details to be tracked. I have been doing this part-time, which means I forget a ton of important details, and have to re-learn them every time I work on any of my plugins.

I would also point out that this is a labour of love task and not a paid job. Which means I don't have the time to spend on this plugin, except when I have time to spare and the desire to spend hours trying to figure out a working combination of IntelliJ Community IDE version and CLion version, which I can use to debug the plugin, and figure out what is really broken or just misconfigured in the dev environment.

All in all, I love the IDEs but hate maintaining plugins for them.

That said, I have uploaded a fix to https://github.com/vsch/CLionArduinoPlugin/blob/master/dist/CLionArduinoPlugin.1.5.9.zip (download and Install plugin from disk, in plugins of the IDE) for the exceptions and fixed a few bugs related to baud rate either being always custom or default for project build settings.

This version also has changes from 1.5.7, brings back the clear button (lost in the chasing the API race) and added an option to clear serial monitor console after reconnecting on build.

I will not make an official build until I can update Arduino-CMake to work with 2022.3 and up.

EasyG0ing1, 2023-03-24:
@vsch - I completely understand each 'pain point' you mentioned concerning maintaining a JetBrains plugin and keeping it current as JetBrains pushes out new releases to their IDEs. The frustrations that you listed can almost be a template for anything that we do with part-time development since coding is a constantly evolving discipline that creates - as you stated - massive complexity that becomes difficult to keep up with. And I fear this paradigm will only get worse as languages evolve and security becomes more complicated to implement among other crucial elements in open source projects.

I do appreciate your efforts here.

It might be a consideration for you to try and wean people away from this project over to PlatformIO. Its CLion plugin seemed pretty stable for 2022.2.3. Still, I haven't checked it with 2022.2.4 or .5. Obviously the folks that actively develop PlatformIO would face the same challenges that you have met with maintaining a CLion plugin. They seem to exclusively prefer Visual Studio - open source edition for MC development using their platform.

Sadly, My hope for a consistent and actively updated option for developing microcontroller code in CLion will continue to be a hit-and-miss option for me. The Arduino IDE is hideous for writing code. Their new version makes some attempts to bring standard IDE features and even offers code completion, but the aesthetics are rather horrid, it feels "klunky" in general, and navigation around the code is severely lacking. CLion is still a much preferred IDE over theirs. And Visual Studio? I think not ... once you get used to using the JetBrains IDEs, it makes it very difficult to use another IDE.

Thank you again - for your efforts here,

Mike Sims