vsch/CLionArduinoPlugin #59: CLion 2022.1 EAP: IllegalStateException: You must not register toolwindow programmatically so early.

indeterminatus, 2022-02-24:
I get an exception on project startup in CLion 2022.1 EAP at least, did not install previous version for regression tests.

My environment:
```
CLion 2022.1 EAP
Build #CL-221.4501.161, built on February 18, 2022
Licensed to CLion EAP user: XXX
Expiration date: XXX
Runtime version: 11.0.14+9-b1993.2 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 10 10.0
GC: G1 Young Generation, G1 Old Generation
Memory: 2000M
Cores: 16
Registry:
    run.processes.with.pty=TRUE

Non-Bundled Plugins:
    idea.plugin.protoeditor (221.4501.155)
    com.jetbrains.plugins.ini4idea (221.4501.155)
    String Manipulation (9.2.0)
    PlantUML integration (5.11.1)
    com.vladsch.clionarduinoplugin (1.5.2)
    intellij.clion.embedded.platformio (221.4501.161)
    izhangzhihao.rainbow.brackets (6.21)
    com.yoheimuta.intellij.plugin.protolint (1.0.1)
    Batch Scripts Support (1.0.13)
```

See this stacktrace for reference:
```
java.lang.IllegalStateException: You must not register toolwindow programmatically so early. Rework code or use ToolWindowManager.invokeLater
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.getToolWindowPaneIfInitialized(ToolWindowManagerImpl.kt:490)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.registerToolWindow(ToolWindowManagerImpl.kt:923)
	at com.jetbrains.rdserver.toolWindow.BackendServerToolWindowManager.registerToolWindow(BackendServerToolWindowManager.kt:89)
	at com.intellij.openapi.wm.ToolWindowManager.registerToolWindow(ToolWindowManager.kt:62)
	at com.vladsch.clionarduinoplugin.serial.SerialMonitorToolWindow.<init>(SerialMonitorToolWindow.java:50)
	at com.vladsch.clionarduinoplugin.serial.SerialProjectComponent.projectOpened(SerialProjectComponent.java:76)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1$1$lambda$1.invoke(ProjectManagerExImpl.kt:503)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1$1$lambda$1.invoke(ProjectManagerExImpl.kt)
	at com.intellij.serviceContainer.ComponentManagerImpl.processInitializedComponents(ComponentManagerImpl.kt:1324)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1$1.run(ProjectManagerExImpl.kt:611)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$executeProcessUnderProgress$12(CoreProgressManager.java:607)
	at com.intellij.openapi.progress.impl.CoreProgressManager.registerIndicatorAndRun(CoreProgressManager.java:682)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computeUnderProgress(CoreProgressManager.java:638)
	at com.intellij.openapi.progress.impl.CoreProgressManager.executeProcessUnderProgress(CoreProgressManager.java:606)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.executeProcessUnderProgress(ProgressManagerImpl.java:60)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1.run(ProjectManagerExImpl.kt:460)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:215)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:22)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:197)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:873)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:511)
	at com.intellij.openapi.application.impl.LaterInvocator$1.run(LaterInvocator.java:96)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:69)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:112)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:42)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:746)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:898)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:746)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:439)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:802)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:438)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:106)
[... 44 lines cut ...]
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:751)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:749)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:748)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:898)
	at com.intellij.ide.IdeEventQueue.dispatchMouseEvent(IdeEventQueue.java:820)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:743)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:439)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:802)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:438)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:119)
	at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:604)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:436)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:873)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:484)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

indeterminatus, 2022-02-24:
Not sure about this plugin's development/maintenance ... the last commits were quite a while ago.

Since this prohibits me from using the serial monitor, I could try to fix it and provide a PR, if there's a chance that this gets this issue resolved :)

coopikoop, 2022-04-29:
I have the same issue, same exception