vsch/CLionArduinoPlugin #62: Immediately after installation: IDE Internal Errors

garykuipers, 2022-08-28:
Instructions were to report this to Vladimir Schenider

stacktrace.txt:
java.lang.IllegalStateException: You must not register toolwindow programmatically so early. Rework code or use ToolWindowManager.invokeLater
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.getToolWindowPaneIfInitialized(ToolWindowManagerImpl.kt:500)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.registerToolWindow(ToolWindowManagerImpl.kt:931)
	at com.jetbrains.rdserver.toolWindow.BackendServerToolWindowManager.registerToolWindow(BackendServerToolWindowManager.kt:102)
	at com.intellij.openapi.wm.ToolWindowManager.registerToolWindow(ToolWindowManager.kt:62)
	at com.vladsch.clionarduinoplugin.serial.SerialMonitorToolWindow.<init>(SerialMonitorToolWindow.java:50)
	at com.vladsch.clionarduinoplugin.serial.SerialProjectComponent.projectOpened(SerialProjectComponent.java:76)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1$2$1.invoke(ProjectManagerExImpl.kt:513)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1$2$1.invoke(ProjectManagerExImpl.kt)
	at com.intellij.serviceContainer.ComponentManagerImpl.processInitializedComponents(ComponentManagerImpl.kt:1324)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1$2.run(ProjectManagerExImpl.kt:645)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$executeProcessUnderProgress$12(CoreProgressManager.java:608)
	at com.intellij.openapi.progress.impl.CoreProgressManager.registerIndicatorAndRun(CoreProgressManager.java:683)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computeUnderProgress(CoreProgressManager.java:639)
	at com.intellij.openapi.progress.impl.CoreProgressManager.executeProcessUnderProgress(CoreProgressManager.java:607)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.executeProcessUnderProgress(ProgressManagerImpl.java:60)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1.run(ProjectManagerExImpl.kt:464)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:513)
	at com.intellij.openapi.application.impl.LaterInvocator$1.run(LaterInvocator.java:95)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:75)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:118)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:42)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:779)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:730)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:749)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:918)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:766)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:450)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:791)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:449)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
	at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:624)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:447)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:493)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)

AzamatKelemetov, 2022-09-10:
Have the same issue right after create a new project - I've got the same stacktrace error.

PerfectlyFineCode, 2022-09-14:
Same issue here, currently on version 1.5.2

oberinspector, 2022-09-24:
Same issue with version 1.5.2. Library option seems to be missing

AdyGCode, 2022-10-21:
v1.5.2 error in CLion 2022.2.4
```text
java.lang.IllegalStateException: You must not register toolwindow programmatically so early. Rework code or use ToolWindowManager.invokeLater
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.getToolWindowPaneIfInitialized(ToolWindowManagerImpl.kt:500)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.registerToolWindow(ToolWindowManagerImpl.kt:930)
	at com.jetbrains.rdserver.toolWindow.BackendServerToolWindowManager.registerToolWindow(BackendServerToolWindowManager.kt:102)
	at com.intellij.openapi.wm.ToolWindowManager.registerToolWindow(ToolWindowManager.kt:62)
	at com.vladsch.clionarduinoplugin.serial.SerialMonitorToolWindow.<init>(SerialMonitorToolWindow.java:50)
	at com.vladsch.clionarduinoplugin.serial.SerialProjectComponent.projectOpened(SerialProjectComponent.java:76)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1$2$1.invoke(ProjectManagerExImpl.kt:513)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1$2$1.invoke(ProjectManagerExImpl.kt)
	at com.intellij.serviceContainer.ComponentManagerImpl.processInitializedComponents(ComponentManagerImpl.kt:1324)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1$2.run(ProjectManagerExImpl.kt:645)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$executeProcessUnderProgress$12(CoreProgressManager.java:608)
	at com.intellij.openapi.progress.impl.CoreProgressManager.registerIndicatorAndRun(CoreProgressManager.java:683)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computeUnderProgress(CoreProgressManager.java:639)
	at com.intellij.openapi.progress.impl.CoreProgressManager.executeProcessUnderProgress(CoreProgressManager.java:607)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.executeProcessUnderProgress(ProgressManagerImpl.java:60)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$useWithScope$lambda$1.run(ProjectManagerExImpl.kt:464)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:513)
	at com.intellij.openapi.application.impl.LaterInvocator$1.run(LaterInvocator.java:95)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:75)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:118)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:42)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:779)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:730)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:749)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:918)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:766)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:450)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:791)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:449)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:113)
	at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:624)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:447)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:493)
	at com.intellij.ide.IdeEventQueue.pumpEventsForHierarchy(IdeEventQueue.java:965)
	at com.intellij.openapi.progress.util.ProgressWindow.lambda$startBlocking$4(ProgressWindow.java:211)
	at com.intellij.openapi.application.impl.ApplicationImpl.runUnlockingIntendedWrite(ApplicationImpl.java:907)
	at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:206)
	at com.intellij.openapi.progress.impl.ProgressRunner.lambda$execFromEDT$6(ProgressRunner.java:329)
	at java.base/java.util.concurrent.CompletableFuture.uniAcceptNow(CompletableFuture.java:757)
	at java.base/java.util.concurrent.CompletableFuture.uniAcceptStage(CompletableFuture.java:735)
	at java.base/java.util.concurrent.CompletableFuture.thenAccept(CompletableFuture.java:2182)
	at com.intellij.openapi.progress.impl.ProgressRunner.execFromEDT(ProgressRunner.java:326)
	at com.intellij.openapi.progress.impl.ProgressRunner.submit(ProgressRunner.java:267)
	at com.intellij.openapi.progress.impl.ProgressRunner.submitAndGet(ProgressRunner.java:193)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
```

jarek7410, 2022-11-17:
java.lang.IllegalStateException: You must not register toolwindow programmatically so early. Rework code or use ToolWindowManager.invokeLater
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.getToolWindowPaneIfInitialized(ToolWindowManagerImpl.kt:490)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.registerToolWindow(ToolWindowManagerImpl.kt:928)
	at com.jetbrains.rdserver.toolWindow.BackendServerToolWindowManager.registerToolWindow(BackendServerToolWindowManager.kt:89)
	at com.intellij.openapi.wm.ToolWindowManager.registerToolWindow(ToolWindowManager.kt:62)
	at com.vladsch.clionarduinoplugin.serial.SerialMonitorToolWindow.<init>(SerialMonitorToolWindow.java:50)
	at com.vladsch.clionarduinoplugin.serial.SerialProjectComponent.projectOpened(SerialProjectComponent.java:76)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1$1$lambda$1.invoke(ProjectManagerExImpl.kt:503)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1$1$lambda$1.invoke(ProjectManagerExImpl.kt)
	at com.intellij.serviceContainer.ComponentManagerImpl.processInitializedComponents(ComponentManagerImpl.kt:1324)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1$1.run(ProjectManagerExImpl.kt:611)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$executeProcessUnderProgress$12(CoreProgressManager.java:608)
	at com.intellij.openapi.progress.impl.CoreProgressManager.registerIndicatorAndRun(CoreProgressManager.java:683)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computeUnderProgress(CoreProgressManager.java:639)
	at com.intellij.openapi.progress.impl.CoreProgressManager.executeProcessUnderProgress(CoreProgressManager.java:607)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.executeProcessUnderProgress(ProgressManagerImpl.java:60)
	at com.intellij.openapi.project.impl.ProjectManagerExImplKt$openProject$$inlined$executeInEdtWithProgress$1.run(ProjectManagerExImpl.kt:460)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:215)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:22)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:197)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:873)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:511)
	at com.intellij.openapi.application.impl.LaterInvocator$1.run(LaterInvocator.java:96)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:69)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:112)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:42)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:776)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:727)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:746)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:898)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:746)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:439)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:803)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:438)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:119)
	at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:604)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:436)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:873)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:484)
	at com.intellij.ide.IdeEventQueue.pumpEventsForHierarchy(IdeEventQueue.java:943)
	at com.intellij.openapi.progress.util.ProgressWindow.lambda$startBlocking$4(ProgressWindow.java:205)
	at com.intellij.openapi.application.impl.ApplicationImpl.runUnlockingIntendedWrite(ApplicationImpl.java:899)
	at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:201)
	at com.intellij.openapi.progress.impl.ProgressRunner.lambda$execFromEDT$6(ProgressRunner.java:321)
	at java.base/java.util.concurrent.CompletableFuture.uniAcceptNow(CompletableFuture.java:753)
	at java.base/java.util.concurrent.CompletableFuture.uniAcceptStage(CompletableFuture.java:731)
	at java.base/java.util.concurrent.CompletableFuture.thenAccept(CompletableFuture.java:2108)
	at com.intellij.openapi.progress.impl.ProgressRunner.execFromEDT(ProgressRunner.java:318)
	at com.intellij.openapi.progress.impl.ProgressRunner.submit(ProgressRunner.java:259)
	at com.intellij.openapi.progress.impl.ProgressRunner.submitAndGet(ProgressRunner.java:185)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)

WaltWilliams, 2022-12-31:
Same issue. I'm using CLion 2022.3.1   

Walt W.

KSDaemon, 2023-01-13:
So seems this plugin is not maintained any more..... Sad..... :(

vsch, 2023-03-11:
@KSDaemon, almost not maintained. It is a pain to spend hours updating the plugin only to keep up with the IDE API changes. I only use CLion for Arduino hobby development and don't have the time to spend on it for every release of the IDE.

I uploaded a new release 1.5.6 which works on CLion 2019.1 to 2022.2. 

It seems CLion 2022.3 has made big changes to their CMake and arduino-cmake does not work. I have zero experience with CMake, I have a perpetual fallback license for CLion 2020.2 and I am not using sufficiently to justify making it work for later versions.

KSDaemon, 2023-03-11:
@vsch I see. Thnx for the clarification! I understand and respect your point! 

FWIW, I managed to set up [Arduino-CMake-Toolchain](https://github.com/KSDaemon/Arduino-CMake-Toolchain), so at least my pet projects compile, all libs are resolved and autocomplete works in CLion. 
Maybe it will be useful to someone.

vsch, 2023-03-12:
@KSDaemon, with what version of CLion did you setup the Arduino-CMake-Toolchain? Or are you using it from the command line only?

If you got it to work with CLion 2022.3 or later, and you don't mind letting me know what you did, then I can update the plugin to do the same.

BTW, JetBrains has a direct link to the plugin before approval: <https://plugins.jetbrains.com/plugin/11301-arduino-support/versions/stable/305519>. I do not know if the link only works for me or if anyone can download it from it.

KSDaemon, 2023-03-12:
@vsch It is working with CLion 2022.3.* 

So, you know, long time ago there was an Arduino Cmake project... Then was born [Arduino-CMake-NG](https://github.com/arduino-cmake/Arduino-CMake-NG)... Which was also forgotten someday... And then [Arduino-CMake-Toolchain](https://github.com/a9183756-gh/Arduino-CMake-Toolchain) got up to the scene :) But, yes, once again it was forgotten and is not maintained for some time. But it became rather popular. So there are a few PRs and Issue discussions. The most actual fork is by @technyon. Can be found here: https://github.com/technyon/Arduino-CMake-Toolchain
But I fell into some problems with it. As I'm on Mac, but other guys are on Windows/Linux. 

I [forked](https://github.com/KSDaemon/Arduino-CMake-Toolchain) that and updated it based on the latest `dev` branch, and picked some PRs from the upstream.

So now it is working with my sketches for ESP8266, ESP32 and even some specific SAMD51-based boards: code resolving, compiling, uploading.

The latest updates in the toolchain include: resolving headers even if they differ from the lib names, recursive inclusion and some other updates.

vsch, 2023-03-13:
@KSDaemon, I was looking at the PR’s and some looked promising and I was planning to take the time to merge them into my fork of Arduino cmake. 

If you have no objections, if you already have it working, I will update mine to yours. It will allow me to actually get it done within a week or two, instead of months. I too am on a Mac so your changes will really help. 

I want to update the plug-in because I use it for my arduino dev and now my daughter is gotten into it. Setting her system up with CLion and the plug-in is a real pain. I’d like to make it smoother and better documented so her classmates can do the same. I think they can benefit from code with me features.

KSDaemon, 2023-03-13:
@vsch Of course not! I'm not a great cmake expert but managed it to work. So would love to help! 
Feel free to reuse my fork. If you have questions — I'll try to help.

KSDaemon, 2023-03-13:
> my daughter is gotten into it
So nice! Mine is a bit young for Arduino, but she loves to play with electric circuit constructor already :)

vsch, 2023-03-25:
@KSDaemon, it is a start. My daughter is 22 years old and has been wiring batteries, LEDs, switches and motors since she was 8. Finally decided to learn electronics and programming on a serious level. On the down side, now that she's serious, I am exhausted from "our" college curriculum. ;)