vsch/CLionArduinoPlugin #64: Error installing plugin 'Arduino Support' (version '1.5.6') on CL-223.8836.42

MJBina, 2023-03-15:
The full error message is:

Error
Plugin 'Arduino Support' (version '1.5.6') is not compatible with the current version of the IDE, because it requires build 222.* or older but the current build is CL-223.8836.42.

Is there an update planned?

vsch, 2023-04-07:
Tried CLion 2023.1.1 on an Arduino project with Arduino Support plugin. Get error on loading:

```
CMake Error at /Applications/CLion.app/Contents/bin/cmake/mac/share/cmake-3.25/Modules/Platform/UnixPaths.cmake:47 (_cmake_record_install_prefix):
  Unknown CMake command "_cmake_record_install_prefix".
Call Stack (most recent call first):
  arduino-cmake/ArduinoToolchain.cmake:23 (include)
  cmake-build-debug/CMakeFiles/3.25.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:59 (project)
```

`CLion.app/Contents/bin/cmake/mac/share/cmake-3.25/Modules/Platform/UnixPaths.cmake:47` Is the culprit, `_cmake_record_install_prefix()` was not in the last working version CLion 2022.2, but is in 2022.3 and 2023.1 and 2023.1.1

Removing the line allows Arduino-CMake projects to work, but you will need to do it manually in the installed app contents.

I don't know if it will work on Windows but since UnixPaths.cmake will not be included in the script it should work.

vsch, 2023-04-07:
I also updated cmake/Platform/Arduino.cmake to scan the included library directory and if it has source files under `src/` to automatically set `RECURSE` for the library. I don't see when not having this done and forcing the user to do it, would be a better option. So I applied the changes. I tripped over this in my project and decided to look into the cmake file.

I also added the option to specify an external Toolchain file, with an option to create from bundled. Again, having the toolchain copied into every project is convenient for sharing the project across systems but is a pain when the toolchain files need updating. Now there is an option to create it locally and new projects will have this toolchain referenced instead of project directory local one.

Did not try it, but specifying another toolchain altogether, like Arduino-CMake-Toolchain might work out of the box.

vsch, 2023-04-19:
@MJBina, @KSDaemon, the dist directory has a new build 1.5.35 <https://github.com/vsch/CLionArduinoPlugin/blob/master/dist/CLionArduinoPlugin-1.5.35.zip>

This one works from 2020.3 to 2023.1, and should work for later versions unless the API changes a lot.

I migrated to gradle build and fixed a bunch of complaints and deprecation warnings. Some I cannot fix without bringing up the minimum version, which I don't think is worth it.

If someone has a Windows machine setup for Arduino development, I would like to know if it works on the latest versions.

List of fixes and changes:

### 1.5.35

* Change: migrate to gradle build
* Change: minimum supported version to `2020.3` for gradle build compatibility with later versions.
* Fix: `ArduinoToolchain.cmake` to work with CLion 2022.3 and 2023.1
* Add: show difference buttons to show diff which allows modification of unbundled files or CMakeLists.txt, in case of `Change Build Options` action.
* Fix: refactor cmake AST command with `CommandArgument` class to encapsulate the value and leading spaces instead of lugging around two separate lists.
* Fix: improve toolchain file path error and info messages
* Fix: revert macro name for `_SRCS` and `_HRDS` to be resolved, with arguments to remain unresolved.
* Fix: resolve white space preservation in regenerated `CMakeLists.txt` file. Only changing text for values which were changed.
* Fix: `CMakeLists.txt` file with mixed project macro name when `PROJECT_NAME` or `CMAKE_PROJECT_NAME` is used to set `_SRCS` and `_HDRS`, while project was set with another macro name, for example `project(${ANOTHER_NAME})`
* Fix: disable diff buttons if the toolchain file is not `ArduinoToolchain.cmake` so that other toolchain files will not be erroneously compared to bundled version.
* Fix: add second diff button for toolchain diff viewing `Platform/Arduino.cmake`
* Fix: allow directories to be selected for toolchain file path.
* Fix: if a directory is selected add `ArduinoToolchain.cmake` file name automatically.
* Add: Toolchain file path option, wth diff and create, to allow not having a `cmake` directory and tool chain files in every project. Also allows to use different toolchain than `ArduinoCmake.cmake`.
* Fix: Arduino.cmake to detect libraries which need RECURSE flag and set it automatically. Determination is made by looking for `src/` subdirectory containing source files. If one is found the library source search is recursed.
* Fix: Arduino.cmake to exclude sources from subdirectories whose name begins with `cmake-build` or `examples`. Otherwise, it went off to use sources from the build directories to determine library dependence.
* Fix: Update `programmers.txt` from Arduino version `1.8.19`
* Fix: apply pull requests from original project for suggested fixes:
  * [Bug fix: Allow linking for sketches]
  * [Let source files depend on included local header files by ntninja · Pull Request #4 · francoiscampbell/arduino-cmake · GitHub]
  * [Add LIBS input property to GENERATE_ARDUINO_EXAMPLE by ryanpfeeley · Pull Request #3 · francoiscampbell/arduino-cmake · GitHub]
* Fix: serial monitor tool window throwing exceptions on closing because some elements were already disposed.
* Fix: serial monitor settings, baud rate was always showing up as custom.
* Fix: project build settings, baud rate was always default, regardless of how it was set in CMakeLists.txt.
* Fix: IDE exceptions about wrong DataContext and missing component value in DataContext for status bar widget
* Fix: lost clear all button in serial monitor console view, deprecated and removed in 2020.1
* Add: Option to clear Serial Console on reconnect after build.

[Add LIBS input property to GENERATE_ARDUINO_EXAMPLE by ryanpfeeley · Pull Request #3 · francoiscampbell/arduino-cmake · GitHub]: https://github.com/francoiscampbell/arduino-cmake/pull/3 "Add LIBS input property to GENERATE_ARDUINO_EXAMPLE by ryanpfeeley · Pull Request #3 · francoiscampbell/arduino-cmake · GitHub"
[Let source files depend on included local header files by ntninja · Pull Request #4 · francoiscampbell/arduino-cmake · GitHub]: https://github.com/francoiscampbell/arduino-cmake/pull/4 "Let source files depend on included local header files by ntninja · Pull Request #4 · francoiscampbell/arduino-cmake · GitHub"
[Bug fix: Allow linking for sketches]: https://github.com/francoiscampbell/arduino-cmake/pull/10

vsch, 2023-04-21:
Pre-Release available. This version factored out serial port comms to a separate plugin, which could be used by multiple plugins without conflicting on the `jssc` implementation. See version notes in the release if interested.

This means the IDE will prompt you to install the `SimpleSerialConnectorService` plugin, when installing Arduino Support.

You can download the pre-release plugin from the assets of the release <https://github.com/vsch/CLionArduinoPlugin/releases/tag/1.5.39> or from this link: <https://github.com/vsch/CLionArduinoPlugin/releases/download/1.5.39/CLionArduinoPlugin-1.5.39.zip> and install the plugin from disk in IDE plugins page.

KSDaemon, 2023-04-28:
Tried to install 1.5.39 release from zip - strangely can not activate the plugin. It installs sucessfully, I can see it in the plugins list, but Enabling/putting a checkmark doesn't make any sense. 

CLion 2023.1.1  / Mac OS

[image]

vsch, 2023-04-29:
Did the IDE prompt you to install SimpleSerialConnectorService plugin? It should have since it is a dependency of Arduino Support plugin.

On my system it prompts to install it and enable it.

You can install the SimpleSerialConnectorService from the marketplace and enable it. Afterwards, you should be able to enable Arduino Support.

KSDaemon, 2023-04-29:
Hmm... No... Just installed it by hands and now all is enabled and looks working! (At least loaded) I'll see how it goes! Thanks for your help!