vsch/CLionArduinoPlugin #70: Board is not selected

highjeans, 2024-04-30:
I am looking to use CLion instead of the Arduino IDE to run code on my ESP32 (It is a custom one from a club). I added the boards.txt file that is used in the Arduino IDE to get the correct board (ESP32-S3 Dev Module). However, whoever I try to make a new sketch, the create button is disabled and I see "Board is not selected" at the bottom (see screenshot). What am I supposed to do? I was able to create a new sketch previously before I downloaded the Legacy Arduino IDE for the Arduino SDK. I am on an M1 Pro mac if that helps.

FYI I need to use Arduino not PlatformIO because the club uses it and I don't want to try anything different.

[image]

highjeans, 2024-04-30:
@vsch Sorry for the ping but I needed urgent help and I've been pulling my hair over this error for a few hours now.

jonnybergdahl, 2024-05-10:
I have this issue as well.

simon88, 2024-05-27:
I have also this issue